espressif/esp-matter #757: ble_gap_adv_set_data failed: Error ESP32:0x05B0001E

PhLuReh, 2023-11-28:
**Describe the bug**
Upon the matter device getting removed from fabric, I implemented to reopen the commissioning window automatically. Mainly for testing purposes, but also valuable for the user. The implementation is getting called, but fails in runtime (please see device logs for more details):
```
I (223324) app_main: Fabric removed successfully
I (223334) chip[EM]: >>> [E:36822i S:32119 M:2494863 (Ack:181750303)] (S) Msg RX from 1:000000000001B669 [0000] --- Type 0001:01 (IM:StatusResponse)
I (223344) chip[EM]: Dropping message on inactive session that does not match an existing exchange
I (223354) chip[EM]: >>> [E:8353r S:32119 M:2494864 (Ack:181750304)] (S) Msg RX from 1:000000000001B669 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (223374) chip[DIS]: Updating services using commissioning mode 1
I (223374) chip[DIS]: Advertise commission parameter vendorID=5186 productID=4369 discriminator=2132/08 cm=1
E (223404) chip[DL]: ble_gap_adv_set_data failed: Error ESP32:0x05B0001E 2132
E (223404) chip[DL]: Configure Adv Data failed: Error ESP32:0x05B0001E
E (223404) chip[DL]: Disabling CHIPoBLE service due to error: Error ESP32:0x05B0001E
I (223414) app_main: Commissioning window opened
```

I did this implementation, where `openCommissioningWindow()` is being called on fabric removed callback:

```
static void WorkHandler(intptr_t context)
{
    chip::CommissioningWindowManager & commissionMgr = chip::Server::GetInstance().GetCommissioningWindowManager();
    constexpr auto kTimeoutSeconds = chip::System::Clock::Seconds16(commissionMgr.MaxCommissioningTimeout());
    if (!commissionMgr.IsCommissioningWindowOpen())
    {
        // having flag chip::CommissioningWindowAdvertisement::kAllSupported should also open BLE
        // TODO test
        CHIP_ERROR err = commissionMgr.OpenBasicCommissioningWindow(kTimeoutSeconds,
                                        chip::CommissioningWindowAdvertisement::kAllSupported);
        if (err != CHIP_NO_ERROR)
        {
            ESP_LOGE(TAG, "Failed to open commissioning window, err:%" CHIP_ERROR_FORMAT, err.Format());
        }
    } else {
        ESP_LOGI(TAG, "commissioning window already opened, not chance to open again");
    }
}

void openCommissioningWindow()
{
    chip::DeviceLayer::PlatformMgr().ScheduleWork(WorkHandler, intptr_t(nullptr));
}
```

**Environment**
- ESP-Matter Commit Id: dd4f34ec98ee1d5467bcfa8f6c885d53de58e50e
- ESP-IDF Commit Id: e088c3766ba440e72268b458a68f27b6e7d63986
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file): [ble_error.txt](https://github.com/espressif/esp-matter/files/13489724/ble_error.txt)
- Host Machine OS: `Linux version 6.1.29-1-MANJARO (builduser@fv-az292-908) (gcc (GCC) 12.2.1 20230201, GNU ld (GNU Binutils) 2.40) #1 SMP PREEMPT_DYNAMIC Wed May 17 14:00:55 UTC 2023`
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
This open issue https://github.com/espressif/esp-matter/issues/715#issuecomment-1790958079 reports a different problem, but still has the same error in its logs.

dhrishi, 2023-11-28:
@PhLuReh The default ESP Matter examples, disable and deinit BLE after commissioning. Can you please check, if you have disabled the config option `CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING` (idf.py menuconfig -> Component config -> CHIP Device Layer -> BLE Options), as you require BLE even after commissioning is complete.

PhLuReh, 2023-11-28:
Ok, this could also be the solution. Is BLE only active if commissioning window is open? or will it be turned on, even if no advertising happens?

I think the problem may be that Wifi provisioning does not get remove after removing fabric. For the developed product this would also be a reasonable solution. Can you provide a proper API for this way?

PhLuReh, 2023-11-29:
> @PhLuReh The default ESP Matter examples, disable and deinit BLE after commissioning. Can you please check, if you have disabled the config option `CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING` (idf.py [...]
I implemented this, and the error now is gone. Also, rerunning tests in TH is working without removing the Wifi provisioning.

If you now tell me, that having BLE active all the time isn't really a thing in terms of security and energy consumption, I would accept this solution. If not, please provide an other way.

wqx6, 2023-11-30:
If you want to open the commissioning window but not to advertise commissionable node over BLE, you can use `commissionMgr.OpenBasicCommissioningWindow(kTimeoutSeconds, chip::CommissioningWindowAdvertisement::kDnssdOnly)`.

dhrishi, 2023-12-01:
>I think the problem may be that Wifi provisioning does not get remove after removing fabric. 

As of now, the spec does not clearly define what is to be done about the network credentials (Wi-Fi or Thread) after the fabric is removed, especially if it is the last fabric. I have raised an [issue](https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/7635) sometime back (status: open). Note that the spec repo is a private repo and you will be able to access it only if you are a CSA member.

dhrishi, 2024-02-19:
Closing this now

closed 2024-02-19 by dhrishi