espressif/esp-matter #758: Build fails when log level set to none. Observed the same issue with esp matter examples as well (Tried with example matter light)

aggaddam, 2023-11-30:
**Describe the bug**
Build fails when the log level is set to none and observed the same issue with esp matter examples as well (Tried with example matter light). Here is the error that occurred during the build process

**Environment**
- ESP-Matter Commit Id: https://github.com/espressif/esp-matter/commit/68e99de70266674ce172459136960bbb5c58dc0e
- ESP-IDF Commit Id:  v5.1.1 Stable Release
- SoC (eg: ESP32 or ESP32-C3): ESP32-C3
- Device Logs (Please attach the log file):
  ```

nclude_esp_phy_override  -lphy  -lbtbb  -lphy  -lbtbb  esp-idf/esp_phy/libesp_phy.a  -u vfs_include_syscalls_impl && :
/home/agaddam/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/esp_wifi/libesp_wifi.a(esp_adapter.c.obj): in function `esp_log_writev_wrapper':
/home/agaddam/Documents/test/li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_wifi/esp32c3/esp_adapter.c:372: undefined reference to `__wrap_esp_log_writev'
/home/agaddam/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/esp_wifi/libesp_wifi.a(esp_adapter.c.obj): in function `esp_log_write_wrapper':
/home/agaddam/Documents/test/li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_wifi/esp32c3/esp_adapter.c:360: undefined reference to `__wrap_esp_log_writev'
/home/agaddam/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/esp_http_client/libesp_http_client.a(esp_http_client.c.obj): in function `esp_http_client_read':
/home/agaddam/Documents/test/li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_http_client/esp_http_client.c:1164: undefined reference to `__wrap_esp_log_write'
/home/agaddam/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/agaddam/Documents/test/li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/esp_http_client/esp_http_client.c:1164: undefined reference to `__wrap_esp_log_write'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/agaddam/Documents/test/li_ac_matter_fw/targets/simplifi/AC/build/log/idf_py_stderr_output_14766 and /home/agaddam/Documents/test/li_ac_matter_fw/targets/simplifi/AC/build/log/idf_py_stdout_output_14766

MAKE FAIL
```
- Host Machine OS: Ubuntu
- Commissioner app and versions if present: NA
- Commissioner's logs if present: NA

**Any additional details**
...

sayondeep, 2023-11-30:
@aggaddam Can you please enable "Use external log wrapper" and try again..
It can be enabled at `Component config -> Diagnostics -> Use external log wrapper`

aggaddam, 2023-11-30:
@sayondeep It worked. Thank you!

Can i always enable this setting Component config -> Diagnostics -> Use external log wrapper irrespective of log level state?

shubhamdp, 2023-12-07:
@aggaddam We have fixed the problem in https://github.com/espressif/esp-insights/commit/2990bb3e931749d0f374df62b58c8fb558c45454 and you can remove the workaround.

I was wrong about the second part where we update connectedhomeip and then esp-matter... I just works without any changes.

Also, please close the issue once you verify it.

dhrishi, 2023-12-08:
Closing. Please re-open in case of any issues

closed 2023-12-08 by dhrishi