espressif/esp-matter #763: Testharness 2.8.1 general problems

svenehlers, 2023-12-05:
We mostly finished developing a window covering using the esp-matter SDK based on matter v1.1.
Currently we are testing the application using the Testharness v2.8.1 and we face significant problems in the workflow because in our view alot of Tests are not designed correctly or necessary test steps are missing and so on.

Does anybody else have significant problems with the Testharness?
It seems like a huge amount of tests are not testing in the correct way and a lot of errors happen even if the application does the correct thing.
Maybe i will get into more Detail later and give some examples. I just wonder if somebody else got a similar experience in general.

VaishaliAvhale, 2023-12-13:
@svenehlers, can you please point me to the test Case/TC-ID where you have observed a discrepancy?

dhrishi, 2024-01-29:
I will close this issue now. If you have any specific tests that are failing, due to the test steps or any other setup issues, please file an issue under connectedhomeip.

closed 2024-01-29 by dhrishi