espressif/esp-matter #765: TC-RR-1.1 test issue

tonyVopo, 2023-12-06:
ESP Matter  SDK：Matter 1.1 release

wqx6, 2023-12-11:
The TC-RR-1.1 could fail when the NVS is not enough. We have increased the partition size of `nvs` in https://github.com/espressif/esp-matter/commit/b292cf047e22c3f0e06a0e0d612410377f3564ee. Could you please increase the partition size and try again?

tonyVopo, 2023-12-12:
@wqx6 Yes, we solved the problem by increasing the size of the NVS partition.  Thx

closed 2023-12-12 by tonyVopo

emesamwong, 2024-03-08:
hi, @wqx6 

My partition for nvs is already 48K but still fail in the TC-RR-1.1 and the same issue as tonyVopo mentioned
**DUT report Failure in KeySetWrite command**

[image]

Do I need the increase the nvs size???

Cert: Matter 1.2
ESP-32S3
esp-matter@2f692ac1
connectedhomeip@8e3d98c0