espressif/esp-matter #768: Adding TagList for Generic Switch endpoints

FHauschild, 2023-12-07:
Hi,

I'm developing a device with multiple switches.
I'm trying to use the TagList to describe the switches with tags like Number.One as described in the new Matter specification (Matter Device Library v1.2 Chapter 6.6.4.2).

How do I add the TagList Attribute to the Descriptor cluster ?
And How do I Write the TagList value to the Attribute?

I'm doing this right now, but it doesn't seem to work.

```
      cluster_t* p_btn1_descriptor_cluster = cluster::get(p_btn_1_endpoint, Descriptor::Id);
      Descriptor::Structs::SemanticTagStruct::Type btn_1_tag;
      btn_1_tag.namespaceID = 0x43;   // Switch Namespace ID
      btn_1_tag.tag         = 0;      // "On""

      esp_matter_attr_val_t btn_1_tag_list_val = esp_matter_invalid(NULL);
      btn_1_tag_list_val = esp_matter_array((uint8_t*)&btn_1_tag, sizeof(btn_1_tag_list_val), 1);
      
      // Add Tags for Switches
      attribute_t* p_test_attribute = NULL;
      p_test_attribute = attribute::create(p_btn1_descriptor_cluster, Descriptor::Attributes::TagList::Id, CLUSTER_FLAG_SERVER, btn_1_tag_list_val);
```

I'm also unable to see the Tags in the Apple Home app or in Google Home.
But I think the TagList is not supported yet.

shripad621git, 2023-12-22:
@FHauschild , thanks for reporting.We will add the support  soon.

shripad621git, 2024-01-12:
@FHauschild ,the fetaure is available to test now.You can try out the feature by using the [generic_switch](https://github.com/espressif/esp-matter/tree/main/examples/generic_switch) example.

dhrishi, 2024-01-15:
Closing this as the support is added