espressif/esp-matter #769: Lighting-app startup error on esp32h2: uncommissionable!

swdcom, 2023-12-08:
**Describe the bug**
Built and flashed the lighting-app example on a esp32h2. When using idf.py monitor an app-task error can be seen, when initial values get set for the lighting-app. Commissioning to a thread network is not possible, just a base router functionality of a FTD remains. All because the AppTask has an error (I think).

**Environment**
- Matter Commit Id: [v1.2.0.0](https://github.com/project-chip/connectedhomeip/tree/v1.2.0.0)
- ESP-IDF Commit Id: [v5.1.1](https://github.com/espressif/esp-idf/tree/v5.1.1)
- SoC: ESP32-H2
- Device Logs: [idf.py_monitor.log](https://github.com/espressif/esp-matter/files/13612766/idf.py_monitor.log) line 144 ff.
- Host Machine OS: docker image of ubuntu : '$ lsb_release -a'
```
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.6 LTS
Release:	20.04
Codename:	focal
```
- Commissioner app and versions if present: chip-tool from matter v1.2.0.0
- Commissioner's logs if present: can do but won't help.

**Any additional details**
Border Router is an OTBR based on Raspberry Pi4 running Bookworm and a nRF 54840 dongle.

Looking at connectedhomeip/example/lighting-app/esp32/main/AppTask.cpp the source of the error is here:
```
void AppTask::UpdateClusterState()
{
    ESP_LOGI(TAG, "Writing to OnOff cluster");
    // write the new on/off value
    EmberAfStatus status = Clusters::OnOff::Attributes::OnOff::Set(kLightEndpointId, AppLED.IsTurnedOn());

    if (status != EMBER_ZCL_STATUS_SUCCESS)
    {
        ESP_LOGE(TAG, "Updating on/off cluster failed: %x", status);
    }

    ESP_LOGI(TAG, "Writing to Current Level cluster");
    status = Clusters::LevelControl::Attributes::CurrentLevel::Set(kLightEndpointId, AppLED.GetLevel());

    if (status != EMBER_ZCL_STATUS_SUCCESS)
    {
        ESP_LOGE(TAG, "Updating level cluster failed: %x", status);
    }
}
```
When "Writing to Current Level cluster" that fails with 0x87 - EMBER_ZCL_STATUS_CONSTRAINT_ERROR

Thinking about this: An OnOff light does not have a level, right?

(Stephan)

swdcom, 2023-12-08:
But mine doesn't even start up right.

shubhamdp, 2023-12-08:
@swdcom Attached logs shows that device is advertising over BLE and is in commissionable mode. This means that you can commission the device.

Whereas if you are concern about the below logs (related to the code snippet that you poste)
```
I (2227) app-task: Writing to Current Level cluster
E (2227) app-task: Updating level cluster failed: 87
```

This is the constraint error because range for current-level is 0-254 and the code is setting it to [255](https://github.com/project-chip/connectedhomeip/blob/master/examples/lighting-app/esp32/main/LEDWidget.cpp#L30). Not sure if that change is intentional or a bug, will analyze and fix it.

swdcom, 2023-12-08:
So do I understand correctly that there is implemented a Level Control Cluster (0x8), that is initialized with 255 by the LEDWidget. But max. allowed level in matter is 254, as stated in matter application cluster specification?

> 1.6.4.2. Lighting Feature
> This feature supports an interface for controlling the level of a light source.
> For the CurrentLevel attribute:
>    A value of 0x00 SHALL NOT be used.
>    A value of 0x01 SHALL indicate the minimum level that can be attained on a device.
>    A value of 0xfe SHALL indicate the maximum level that can be attained on a device.
>    A value of null SHALL represent an undefined value.
>    All other values are application specific gradations from the minimum to the maximum level.

What about the lowest level? Is it possible to demand level 0 via matter, get it set in the LEDWidget::SetBrightness(…) but then fail on persisting it in the attribute storage?

(Stephan)

swdcom, 2023-12-11:
For the record: That the example wasn't commissionable was most likely my fault. I did not understand the srp errors in the log about "name already used". It seems that in a previous commissioning attempt I had used the same node id, which was still present to chip-tool. I used a 'new' node id which made commissioning of the example work again. Just plain stupid coincidence.

(Stephan)

dhrishi, 2023-12-22:
Closing this

closed 2023-12-22 by dhrishi