espressif/esp-matter #770: Unable to connect Zigbee light example to Zigbee Matter Bridge on Zigbee Gateway board (StackOverflow?)

DanTup, 2023-12-12:
I flashed the Zigbee on_off_light and on_off_switches to ESP32-H2 devkits, and everthing worked fine. I can toggle the light on the first H2 (light) by pressing the Boot button on the second H2 (switch).

Now I'd like to try and connect them to my WiFi network and have them show through Alexa/Echo by using the Matter Zigbee Bridge.

I'm using ESP IDF release 5.1.2 and 72b0f26 from thie repository (I couldn't find any obvious docs on which versions I should use, but the [sdkconfig.defaults.zb_gw_board](https://github.com/espressif/esp-matter/blob/main/examples/zigbee_bridge/sdkconfig.defaults.zb_gw_board) file only exists in recent commits). I have the Zigbee Gateway board (which I've previously used as a TBR successfully). I built and flashed the zigbee_rcp from the ESP-IDF repo to the H2 side of the gateway board, and then flashed the zigbee_bridge from this repo (after selecting the board type in `menuconfig`) to the S3 side.

I was able to add the bridge as a Matter device in the Amazon app using the QR code from [this page](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html) and it shows up as a hub.

However, it's not clear to me whether the light on the standalone H2 is joining the Zigbee network. There's no obvious output in the monitor of the gateway board, and if I try to search for it through the alexa app nothing shows up. However, there are some errors (StackOverflows) in the output that perhaps are related (it also says "CORRUPTED"!):

```
***ERROR*** A stack overflow in task console has been detected.

Backtrace: 0x40375a5a:0x3fcb3790 0x4037f401:0x3fcb37b0 0x40381c8a:0x3fcb37d0 0x40380739:0x3fcb3850 0x40381d94:0x3fcb3880 0x40381d8a:0x00000800 |<-CORRUPTED
0x40375a5a: panic_abort at /Users/danny/Dev/Microcontrollers/esp-idf/components/esp_system/panic.c:472

0x4037f401: esp_system_abort at /Users/danny/Dev/Microcontrollers/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x40381c8a: vApplicationStackOverflowHook at /Users/danny/Dev/Microcontrollers/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:581

0x40380739: vTaskSwitchContext at /Users/danny/Dev/Microcontrollers/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3841

0x40381d94: _frxt_dispatch at /Users/danny/Dev/Microcontrollers/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:450

0x40381d8a: _frxt_int_exit at /Users/danny/Dev/Microcontrollers/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:245
```

The full log is here: https://gist.github.com/DanTup/483d5c030e02db441d7039cc5a990bb1

Any help/advice would be appreciated!

DanTup, 2023-12-16:
I switched to using the 5.2 branch for the RCP and the crash I saw when running `help` has gone away, but I I still can't connect a zigbee device. In the output I see:

```
Zigbee stack initialized
```

But I never see `Start network formation`, `Network steering started` or `Zigbee rcp device booted`.

Everything works fine with the HA_on_off_switch example on an H2, so my feeling is that maybe the issue is somewhere between the S3/H2 on the gateway board. The board definitely works as an open border router though, so I don't think it's a hardware issue.

I'm not sure how to debug, but perhaps a good start would be knowing exactly which version of the rcp from the esp-idf repo I should pair which which version of the zigbee_bridge from the esp-matter repo (because right now there doesn't seem to be any docs on this, and I don't know if this is important, because I'm not familiar with the protocol used for the two to communicate).

DanTup, 2023-12-17:
Ok, I got this working. I noticed the versions of the Zigbee components was quite old:

https://github.com/espressif/esp-matter/blob/72b0f2632690c4b2917e6ae7b359297f593da1cb/examples/zigbee_bridge/main/idf_component.yml#L3-L4

Running `idf.py update-dependencies` didn't seem to update them (I only got 0.5.4 and 0.7.4) so I manually updated them to:

```
  espressif/esp-zboss-lib: "~1.0.6"
  espressif/esp-zigbee-lib: "~1.0.6"
```

And now everything works - I see the familiar "Network steering" output, and my devices join the network. I can now control some test Zigbee switches via Matter in the Echo app :-)

dhrishi, 2023-12-18:
@DanTup Thanks for the issue as well as the solution :-D We will check and update the file as required. I will keep this open until the change is synced to Github.

shubhamdp, 2024-01-03:
@DanTup Component versions were updated in commit and that would fix this issue https://github.com/espressif/esp-matter/commit/6fe92f9f48e12995a78e5a5fe46085228bb29738

closed 2024-01-03 by dhrishi