espressif/esp-matter #771: Apple Home - Commissions, then No Response - `CASE Session establishment failed: 32`

mriksman, 2023-12-14:
**Describe the bug**
Months ago I built a development environment (iOS 16.x and ESP-IDF 4.4 at the time) and loaded the Light example. It worked fine.

I have come back to the project and rebuilt the environment. Running VSCode with ESP-IDF checkout `6b1f40b9bf`, used the ESP-IDF extension to install `esp-matter` and loaded the Light example. It commissioned OK, but shows No Response in the Apple Home App. The terminal shows the following, repeating occassionally.
```
I (571069) chip[EM]: Retransmitting MessageCounter:249858083 on exchange 6847r Send Cnt 4
I (571249) chip[EM]: Retransmitting MessageCounter:249858084 on exchange 27599r Send Cnt 1
I (571649) chip[EM]: Retransmitting MessageCounter:249858084 on exchange 27599r Send Cnt 2
I (572219) chip[EM]: Retransmitting MessageCounter:249858084 on exchange 27599r Send Cnt 3
E (572469) chip[EM]: Failed to Send CHIP MessageCounter:249858083 on exchange 6847r sendCount: 4 max retries: 4
I (573069) chip[EM]: Retransmitting MessageCounter:249858084 on exchange 27599r Send Cnt 4
E (574539) chip[EM]: Failed to Send CHIP MessageCounter:249858084 on exchange 27599r sendCount: 4 max retries: 4
E (578829) chip[SC]: CASESession timed out while waiting for a response from the peer. Current state was 2
E (578839) chip[IN]: CASE Session establishment failed: 32
```

**Environment**
- ESP-Matter Commit Id: `72b0f2632690c4b2917e6ae7b359297f593da1cb`
- ESP-IDF Commit Id: `6b1f40b9bf`
- SoC (eg: ESP32 or ESP32-C3): ESP23
- Device Logs (Please attach the log file):
- Host Machine OS: Ubuntu 22.04
- Commissioner app and versions if present:
- Commissioner's logs if present:

VaishaliAvhale, 2023-12-14:
Please share the complete DUT logs in the file and provide details about the Apple Home (iPad or iOS version and HomePod version). 
Which DUT are you using, esp32c3, or esp32?

mriksman, 2023-12-16:
Attached logs. 

ESP32-S AiThinker.
iOS 17.2
Multiple HomePods (17.1.1)/Apple TVs (17.1)

Strangest thing - the first log (`LogFail.txt`) showed the commissioning failing. That was new. Previously it commissioned and then showed 'No Response'.
[LogFail.txt](https://github.com/espressif/esp-matter/files/13691514/LogFail.txt)
So I factory reset (shown in the logs) and tried again. This time, it commissioned and appears working in the Apple Home App (`Logs.txt`). 
[Logs.txt](https://github.com/espressif/esp-matter/files/13691513/Logs.txt)
It's concerning that I am getting different results each time I try...

mriksman, 2023-12-16:
I went away, came back, unplugged the USB, plugged it back in, and now I am seeing the 'No Response' issue. 
Attached log.
[LogLaterRestart.txt](https://github.com/espressif/esp-matter/files/13692355/LogLaterRestart.txt)

mriksman, 2023-12-17:
At some point during the night, they became unresponsive; showing 'No Response' in Apple Home. See attached logs.
[LogFailedOvernight.txt](https://github.com/espressif/esp-matter/files/13694391/LogFailedOvernight.txt)

shripad621git, 2023-12-20:
@mriksman ,I think the discussion over [here](https://github.com/espressif/esp-matter/issues/431) answers your query.

mriksman, 2023-12-20:
Perhaps. I created a new 'Home', moved a HomePod mini over to it, and added the Matter Accessory. So far it has been rock solid/stable.

Perhaps I had/have an issue with my current/old Home? I have noticed other strange things happen besides the above. I might move everything over to the new Home, and monitor it.

If there is nothing obviously wrong in my logs above, then this issue can be closed.

closed 2024-01-03 by dhrishi