espressif/esp-matter #772: ImportError: cannot import name 'OneStyleAndTextTuple' from 'prompt_toolkit.formatted_text

mertmzk, 2023-12-14:
Hello followed the instruction from the website: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#development-setup

and when installing to matter getting this error. Any idea how this could be fixed.

**Environment**
- ESP-Matter Commit Id: 72b0f2632690c4b2917e6ae7b359297f593da1cb
- ESP-IDF Commit Id: 6b1f40b9bfb91ec82fab4a60e5bfb4ca0c9b062f
- SoC ESP32
- Device Logs (Please attach the log file):
- Host Machine OS: MAC 13.3.1

Thanks

mertmzk, 2023-12-14:
Fix:

[image]

dhrishi, 2023-12-19:
Fixed with the latest main branch

closed 2023-12-19 by dhrishi