espressif/esp-matter #773: Thermostat  setting temperature error

MaplestoryAlen, 2023-12-15:
**Describe the bug**
    Using the chip tool to send heating can only be done at 22-24 degrees Celsius and cooling can only be done at 23-26 degrees Celsius. Failure to do so will result in failure
 
The attribute configuration is as follows：
 
  namespace heating {
typedef struct config {
  
   int16_t occupied_heating_setpoint; 
   int16_t max_heat_setpoint_limit; 
   int16_t min_heat_setpoint_limit; 
   int16_t abs_max_heat_setpoint_limit;
   int16_t abs_min_heat_setpoint_limit; 
  config():occupied_heating_setpoint(2000),max_heat_setpoint_limit(3500),min_heat_setpoint_limit(2200),abs_max_heat_setpoint_limit(3500),abs_min_heat_setpoint_limit(2200){}
   	
} config_t;

uint32_t get_id();
esp_err_t add(cluster_t *cluster, config_t *config);
} /* heating */

namespace cooling {
typedef struct config {

   int16_t occupied_cooling_setpoint;
   int16_t max_cool_setpoint_limit; 
   int16_t min_cool_setpoint_limit; 
   int16_t abs_max_cool_setpoint_limit;
   int16_t abs_min_cool_setpoint_limit;
	
   config (): occupied_cooling_setpoint(2600),max_cool_setpoint_limit(3000),min_cool_setpoint_limit(1600),abs_max_cool_setpoint_limit(3000),abs_min_cool_setpoint_limit(1600) {}

} config_t;

uint32_t get_id();
esp_err_t add(cluster_t *cluster, config_t *config);
} /* cooling */

chip_tool command :
./chip-tool thermostat write unoccupied-cooling-setpoint  2200 0x12 1
return:
 CHIP:TOO: Run command failure: IM Error 0x00000587: General error: 0x87 (CONSTRAINT_ERROR)

**Environment**
- ESP-Matter Commit Id:V1.2
- ESP-IDF Commit Id :ESP-IDF v5.1.1
- SoC : ESP32S3
- Device Logs (Please attach the log file):
- Host Machine OS:
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

jadhavrohit924, 2023-12-19:
@MaplestoryAlen I was able to set occupied_heating_setpoint/occupied_cooling_setpoint within its min/max limits, could you please check again? Or please share whole patch with me so that I can check what is wrong.

MaplestoryAlen, 2023-12-19:
> @MaplestoryAlen I was able to set occupied_heating_setpoint/occupied_cooling_setpoint within its min/max limits, could you please check again? Or please share whole patch with me so that I can check what is wrong.

From the logs, it appears that the temperature setting has been successful, but the chip tool returned an error code of 0x87

matter version：v1.2 
chip-tool command: ./chip-tool thermostat write occupied-cooling-setpoint  2200 0x12 1
return: CHIP:TOO: Run command failure: IM Error 0x00000587: General error: 0x87 (CONSTRAINT_ERROR)

log:

 I (490124) chip[EM]: >>> [E:58782r S:0 M:94660592] (U) Msg RX from 0:F64840654B6DC60F [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (490144) chip[IN]: CASE Server received Sigma1 message . Starting handshake. EC 0x3fca5e38
I (490144) chip[EM]: <<< [E:58782r S:0 M:32257736 (Ack:94660592)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::1F7B:4D17:A0AA:80E4%st1]:59846] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (490174) chip[SC]: Received Sigma1 msg
I (490174) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0x0000000000000012
I (490424) chip[EM]: <<< [E:58782r S:0 M:32257737 (Ack:94660592)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::1F7B:4D17:A0AA:80E4%st1]:59846] --- Type 0000:31 (SecureChannel:CASE_Sigma2)
I (490444) chip[SC]: Sent Sigma2 msg
I (490454) chip[EM]: >>> [E:58782r S:0 M:94660593 (Ack:32257737)] (U) Msg RX from 0:F64840654B6DC60F [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3)
I (490464) chip[EM]: <<< [E:58782r S:0 M:32257738 (Ack:94660593)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::1F7B:4D17:A0AA:80E4%st1]:59846] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (490484) chip[SC]: Received Sigma3 msg
E (490494) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
I (491084) chip[EM]: <<< [E:58782r S:0 M:32257739 (Ack:94660593)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::1F7B:4D17:A0AA:80E4%st1]:59846] --- Type 0000:40 (SecureChannel:StatusReport)
I (491104) chip[SC]: SecureSession[0x3fca47a0, LSID:47532]: State change 'kEstablishing' --> 'kActive'
I (491104) chip[IN]: CASE Session established to peer: <000000000001B669, 1>
I (491124) chip[EM]: >>> [E:58783r S:47532 M:5478180] (S) Msg RX from 1:000000000001B669 [5A8B] --- Type 0001:06 (IM:WriteRequest)
I (491134) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x0000FFFC is 63 **********
I (491164) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000011 is 2200 **********
I (491194) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x0000FFFC is 63 **********
I (491214) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000019 is 25 **********
I (491224) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000005 is 1600 **********
I (491254) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000006 is 3000 **********
I (491264) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000017 is 1600 **********
I (491274) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000018 is 3000 **********
I (491304) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000003 is 2200 **********
I (491314) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000004 is 3500 **********
I (491334) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000015 is 2200 **********
I (491354) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000016 is 3500 **********
I (491364) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000011 is 2600 **********
I (491384) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000012 is 2000 **********
I (491404) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000013 is 2800 **********
I (491414) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000014 is 2000 **********
I (491424) chip[EM]: <<< [E:58783r S:47532 M:246978170 (Ack:5478180)] (S) Msg TX to 1:000000000001B669 [5A8B] [UDP:[FE80::1F7B:4D17:A0AA:80E4%st1]:59846] --- Type 0001:07 (IM:WriteResponse)
I (491454) chip[EM]: >>> [E:58782r S:0 M:94660594 (Ack:32257739)] (U) Msg RX from 0:F64840654B6DC60F [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (491474) chip[EM]: >>> [E:58783r S:47532 M:5478181 (Ack:246978170)] (S) Msg RX from 1:000000000001B669 [5A8B] --- Type 0000:10 (SecureChannel:StandaloneAck)

chip-tool command: ./chip-tool thermostat write  occupied-heating-setpoint  2600 0x12 1
return: CHIP:TOO: Run command failure: IM Error 0x00000587: General error: 0x87 (CONSTRAINT_ERROR)

log:
> I (703124) chip[EM]: >>> [E:7616r S:0 M:56636807] (U) Msg RX from 0:7BFBE5B406BD9312 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (703144) chip[IN]: CASE Server received Sigma1 message . Starting handshake. EC 0x3fca5e38
[... cut ...]
I (704424) chip[EM]: <<< [E:7617r S:47535 M:261578834 (Ack:146150301)] (S) Msg TX to 1:000000000001B669 [5A8B] [UDP:[FE80::1F7B:4D17:A0AA:80E4%st1]:33822] --- Type 0001:07 (IM:WriteResponse)
I (704454) chip[EM]: >>> [E:7616r S:0 M:56636809 (Ack:32257751)] (U) Msg RX from 0:7BFBE5B406BD9312 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (704464) chip[EM]: >>> [E:7617r S:47535 M:146150302 (Ack:261578834)] (S) Msg RX from 1:000000000001B669 [5A8B] --- Type 0000:10 (SecureChannel:StandaloneAck)

jadhavrohit924, 2023-12-19:
From logs, there is failure in setting attribute values. I would suggest  clean build the app, erase_flash, flash and rebuild the chip-tool. Also could you please share the whole patch that you are trying out?

MaplestoryAlen, 2023-12-19:
> From logs, there is failure in setting attribute values. I would suggest clean build the app, erase_flash, flash and rebuild the chip-tool. Also could you please share the whole patch that you are trying out?

I have tried deleting the build folder and using the clean and fullclrun commands, as well as refactoring the project, but the same temperature setting still resulted in errors
Github downloaded the mater release v1.2 branch，But I don't know how to use any commands to view my mater version，so do I need to update the latest mater version?

MaplestoryAlen, 2023-12-19:
> Just use the latest esp-matter, it should work. If you can please attach a patch that you are trying out.
 The attachment is as follows
[esp_matter_feature.txt](https://github.com/espressif/esp-matter/files/13713582/esp_matter_feature.txt)
[esp_matter_feature_h.txt](https://github.com/espressif/esp-matter/files/13713584/esp_matter_feature_h.txt)
[main.txt](https://github.com/espressif/esp-matter/files/13713587/main.txt)

jadhavrohit924, 2023-12-20:
@MaplestoryAlen Are you on latest esp-matter commit? Did you did clean build and erase flash on device? If yes, could you please attach the chip tool logs of reading the occupied-heating-setpoint/occupied-cooling-setpoint and its corresponding min/max attribute values?

MaplestoryAlen, 2023-12-20:
> @MaplestoryAlen Are you on latest esp-matter commit? Did you did clean build and erase flash on device? If yes, could you please attach the chip tool logs of reading the occupied-heating-setpoint/occupied-cooling-setpoint and its corresponding min/max attribute values?

The logs I read have already been mentioned in the reply above. I am using a version that was around 2 months ago, so I am updating the latest version. I wonder if it can solve the above problem?

log：
I (704364) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000011 is 2600 **********
I (704384) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000012 is 2000 **********

jadhavrohit924, 2023-12-20:
Please read the min/max attributes values too!!

MaplestoryAlen, 2023-12-21:
> Please read the min/max attributes values too!!

chip-tool:
send:   ./chip-tool thermostat read min-heat-setpoint-limit 0x12 1
retun:
[1703124869.695409][1028346:1028349] CHIP:DMG: ReportDataMessage =
[1703124869.695499][1028346:1028349] CHIP:DMG: {
[1703124869.695648][1028346:1028349] CHIP:DMG: 	AttributeReportIBs =
[1703124869.695738][1028346:1028349] CHIP:DMG: 	[
[1703124869.695799][1028346:1028349] CHIP:DMG: 		AttributeReportIB =
[1703124869.695864][1028346:1028349] CHIP:DMG: 		{
[1703124869.695925][1028346:1028349] CHIP:DMG: 			AttributeDataIB =
[1703124869.695988][1028346:1028349] CHIP:DMG: 			{
[1703124869.696071][1028346:1028349] CHIP:DMG: 				DataVersion = 0x34a69bba,
[1703124869.696132][1028346:1028349] CHIP:DMG: 				AttributePathIB =
[1703124869.696211][1028346:1028349] CHIP:DMG: 				{
[1703124869.696323][1028346:1028349] CHIP:DMG: 					Endpoint = 0x1,
[1703124869.696423][1028346:1028349] CHIP:DMG: 					Cluster = 0x201,
[1703124869.696486][1028346:1028349] CHIP:DMG: 					Attribute = 0x0000_0015,
[1703124869.696547][1028346:1028349] CHIP:DMG: 				}
[1703124869.696611][1028346:1028349] CHIP:DMG: 					
[1703124869.696702][1028346:1028349] CHIP:DMG: 				Data = 2200, 
[1703124869.696790][1028346:1028349] CHIP:DMG: 			},
[1703124869.696897][1028346:1028349] CHIP:DMG: 			
[1703124869.696991][1028346:1028349] CHIP:DMG: 		},
[1703124869.697072][1028346:1028349] CHIP:DMG: 		
[1703124869.697173][1028346:1028349] CHIP:DMG: 	],
[1703124869.697250][1028346:1028349] CHIP:DMG: 	
[1703124869.697338][1028346:1028349] CHIP:DMG: 	SuppressResponse = true, 
[1703124869.697401][1028346:1028349] CHIP:DMG: 	InteractionModelRevision = 11
[1703124869.698380][1028346:1028349] CHIP:DMG: }
[1703124869.698604][1028346:1028349] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0201 Attribute 0x0000_0015 DataVersion: 883334074
[1703124869.698779][1028346:1028349] CHIP:TOO:   MinHeatSetpointLimit: 2200

esp32s3 log:

> I (554704) chip[EM]: >>> [E:63415r S:0 M:15978984] (U) Msg RX from 0:4A8657CC7ADB9341 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (554714) chip[IN]: CASE Server received Sigma1 message . Starting handshake. EC 0x3fca5e38
I (554724) chip[EM]: <<< [E:63415r S:0 M:134840396 (Ack:15978984)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::1F7B:4D17:A0AA:80E4%st1]:39317] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (554744) chip[SC]: Received Sigma1 msg
I (554754) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0x0000000000000012
I (555004) chip[EM]: <<< [E:63415r S:0 M:134840397 (Ack:15978984)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::1F7B:4D17:A0AA:80E4%st1]:39317] --- Type 0000:31 (SecureChannel:CASE_Sigma2)
I (555014) chip[SC]: Sent Sigma2 msg
I (555034) chip[EM]: >>> [E:63415r S:0 M:15978985 (Ack:134840397)] (U) Msg RX from 0:4A8657CC7ADB9341 [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3)
I (555054) chip[EM]: <<< [E:63415r S:0 M:134840398 (Ack:15978985)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::1F7B:4D17:A0AA:80E4%st1]:39317] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (555064) chip[SC]: Received Sigma3 msg
E (555074) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
I (555664) chip[EM]: <<< [E:63415r S:0 M:134840399 (Ack:15978985)] (U) Msg TX to 0:0000000000000000 [0000] [UDP:[FE80::1F7B:4D17:A0AA:80E4%st1]:39317] --- Type 0000:40 (SecureChannel:StatusReport)
I (555684) chip[SC]: SecureSession[0x3fca49e0, LSID:26486]: State change 'kEstablishing' --> 'kActive'
I (555684) chip[IN]: CASE Session established to peer: <000000000001B669, 1>
I (555704) chip[EM]: >>> [E:63416r S:26486 M:22081547] (S) Msg RX from 1:000000000001B669 [5A8B] --- Type 0001:02 (IM:ReadRequest)
I (555714) chip[EM]: >>> [E:63415r S:0 M:15978986 (Ack:134840399)] (U) Msg RX from 0:4A8657CC7ADB9341 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (555744) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x0000FFFC is 63 **********
I (555754) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000015 is 2200 **********
I (555764) chip[EM]: <<< [E:63416r S:26486 M:206395114 (Ack:22081547)] (S) Msg TX to 1:000000000001B669 [5A8B] [UDP:[FE80::1F7B:4D17:A0AA:80E4%st1]:39317] --- Type 0001:05 (IM:ReportData)
I (555814) chip[EM]: >>> [E:63416r S:26486 M:22081548 (Ack:206395114)] (S) Msg RX from 1:000000000001B669 [5A8B] --- Type 0000:10 (SecureChannel:StandaloneAck)

chip-tool:
send :  ./chip-tool thermostat read max-heat-setpoint-limit 0x12 1
return:
[1703125140.095837][1029318:1029320] CHIP:DMG: ReportDataMessage =
[1703125140.095966][1029318:1029320] CHIP:DMG: {
[1703125140.096027][1029318:1029320] CHIP:DMG: 	AttributeReportIBs =
[1703125140.096155][1029318:1029320] CHIP:DMG: 	[
[1703125140.096223][1029318:1029320] CHIP:DMG: 		AttributeReportIB =
[1703125140.096289][1029318:1029320] CHIP:DMG: 		{
[1703125140.096369][1029318:1029320] CHIP:DMG: 			AttributeDataIB =
[1703125140.096434][1029318:1029320] CHIP:DMG: 			{
[1703125140.096499][1029318:1029320] CHIP:DMG: 				DataVersion = 0x34a69bba,
[1703125140.096560][1029318:1029320] CHIP:DMG: 				AttributePathIB =
[... 15 lines cut ...]
[1703125140.097995][1029318:1029320] CHIP:DMG: }
[1703125140.098165][1029318:1029320] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0201 Attribute 0x0000_0016 DataVersion: 883334074
[1703125140.098298][1029318:1029320] CHIP:TOO:   MaxHeatSetpointLimit: 3500

eps32s3 log:
[... cut ...]
I (32274) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x0000FFFC is 63 **********
I (32284) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000016 is 3500 **********
I (32294) chip[EM]: <<< [E:48989r S:16392 M:231062431 (Ack:124276753)] (S) Msg TX to 1:000000000001B669 [5A8B] [UDP:[FE80::1F7B:4D17:A0AA:80E4%st1]:53230] --- Type 0001:05 (IM:ReportData)
I (32324) chip[EM]: >>> [E:48989r S:16392 M:124276754 (Ack:231062431)] (S) Msg RX from 1:000000000001B669 [5A8B] --- Type 0000:10 (SecureChannel:StandaloneAck)

jadhavrohit924, 2023-12-22:
@MaplestoryAlen I have understood the issue, since you have enabled all the feature of the thermostat cluster along with AUTO feature. And the spec says when you have AUTO feature enabled then `OccupiedHeatingSetpoint <= OccupiedCoolingSetpoint - MinSetpointDeadBand`  and `OccupiedCoolingSetpoint <= OccupiedHeatingSetpoint - MinSetpointDeadBand`. So that you can set OccupiedHeatingSetpoint from MinHeatSetipointLimit (22 degree C) to value(23.5 degree C) at which above condition is satisfied. 

Eg, RequestedOccupiedHeatingSetpoint = 30, OccupiedCoolingSetpoint = 26, and MinSetpointDeadBand = 2.5
So, for above values `OccupiedHeatingSetpoint <= OccupiedCoolingSetpoint - MinSetpointDeadBand` condition fails that why you are getting CONSTRAIN_ERROR.

Ref code link: https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/thermostat-server/thermostat-server.cpp#L169

MaplestoryAlen, 2023-12-25:
> @MaplestoryAlen I have understood the issue, since you have enabled all the feature of the thermostat cluster along with AUTO feature. And the spec says when you have AUTO feature enabled then [...]

This question may not be like this, please see my test below
in heating mode（system-mode = 4), max_heat_setpoint_limit = 3500，max_heat_setpoint_limit = 2200,
write occupied-heating-setpoint = 2600, return error : 0x87 (CONSTRAINT_ERROR),
the temperature values I wrote are all within the range, but there will still be errors !

1. read system-mode
chip-tool command:
 ./chip-tool thermostat read system-mode 0x12 1
return:
[1703484756.633964][14840:14842] CHIP:DMG: ReportDataMessage =
[1703484756.634032][14840:14842] CHIP:DMG: {
[1703484756.634091][14840:14842] CHIP:DMG: 	AttributeReportIBs =
[1703484756.634154][14840:14842] CHIP:DMG: 	[
[1703484756.634213][14840:14842] CHIP:DMG: 		AttributeReportIB =
[1703484756.634276][14840:14842] CHIP:DMG: 		{
[1703484756.634335][14840:14842] CHIP:DMG: 			AttributeDataIB =
[1703484756.634397][14840:14842] CHIP:DMG: 			{
[1703484756.634459][14840:14842] CHIP:DMG: 				DataVersion = 0xae6873f6,
[1703484756.634532][14840:14842] CHIP:DMG: 				AttributePathIB =
[1703484756.634593][14840:14842] CHIP:DMG: 				{
[1703484756.634656][14840:14842] CHIP:DMG: 					Endpoint = 0x1,
[1703484756.634718][14840:14842] CHIP:DMG: 					Cluster = 0x201,
[1703484756.634815][14840:14842] CHIP:DMG: 					Attribute = 0x0000_001C,
[1703484756.634875][14840:14842] CHIP:DMG: 				}
[1703484756.634978][14840:14842] CHIP:DMG: 					
[1703484756.635040][14840:14842] CHIP:DMG: 				Data = 4, 
[1703484756.635115][14840:14842] CHIP:DMG: 			},
[1703484756.635177][14840:14842] CHIP:DMG: 			
[1703484756.635250][14840:14842] CHIP:DMG: 		},
[1703484756.635314][14840:14842] CHIP:DMG: 		
[1703484756.635425][14840:14842] CHIP:DMG: 	],
[1703484756.635501][14840:14842] CHIP:DMG: 	
[1703484756.635563][14840:14842] CHIP:DMG: 	SuppressResponse = true, 
[1703484756.635657][14840:14842] CHIP:DMG: 	InteractionModelRevision = 11
[1703484756.635717][14840:14842] CHIP:DMG: }
[1703484756.635868][14840:14842] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0201 Attribute 0x0000_001C DataVersion: 2926081014
[1703484756.635988][14840:14842] CHIP:TOO:   SystemMode: 4

2. read min-heat-setpoint-limit
./chip-tool thermostat read min-heat-setpoint-limit 0x12 1
rerun:
[1703484857.507915][14940:14942] CHIP:DMG: ReportDataMessage =
[1703484857.507994][14940:14942] CHIP:DMG: {
[1703484857.508053][14940:14942] CHIP:DMG: 	AttributeReportIBs =
[1703484857.508153][14940:14942] CHIP:DMG: 	[
[1703484857.508212][14940:14942] CHIP:DMG: 		AttributeReportIB =
[1703484857.508275][14940:14942] CHIP:DMG: 		{
[1703484857.508334][14940:14942] CHIP:DMG: 			AttributeDataIB =
[1703484857.508395][14940:14942] CHIP:DMG: 			{
[1703484857.508456][14940:14942] CHIP:DMG: 				DataVersion = 0xae6873f6,
[1703484857.508516][14940:14942] CHIP:DMG: 				AttributePathIB =
[1703484857.508577][14940:14942] CHIP:DMG: 				{
[1703484857.508639][14940:14942] CHIP:DMG: 					Endpoint = 0x1,
[1703484857.508702][14940:14942] CHIP:DMG: 					Cluster = 0x201,
[1703484857.508763][14940:14942] CHIP:DMG: 					Attribute = 0x0000_0015,
[1703484857.508838][14940:14942] CHIP:DMG: 				}
[1703484857.509081][14940:14942] CHIP:DMG: 					
[1703484857.509177][14940:14942] CHIP:DMG: 				Data = 2200, 
[1703484857.509237][14940:14942] CHIP:DMG: 			},
[1703484857.509299][14940:14942] CHIP:DMG: 			
[1703484857.509359][14940:14942] CHIP:DMG: 		},
[1703484857.509421][14940:14942] CHIP:DMG: 		
[1703484857.509519][14940:14942] CHIP:DMG: 	],
[1703484857.509667][14940:14942] CHIP:DMG: 	
[1703484857.509767][14940:14942] CHIP:DMG: 	SuppressResponse = true, 
[1703484857.509831][14940:14942] CHIP:DMG: 	InteractionModelRevision = 11
[1703484857.509904][14940:14942] CHIP:DMG: }
[1703484857.510064][14940:14942] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0201 Attribute 0x0000_0015 DataVersion: 2926081014
[... 38 lines cut ...]
[1703485054.613216][15256:15258] CHIP:DMG: {
[1703485054.613259][15256:15258] CHIP:DMG: 	AttributeStatusIBs =
[1703485054.613323][15256:15258] CHIP:DMG: 	[
[1703485054.613369][15256:15258] CHIP:DMG: 		AttributeStatusIB =
[1703485054.613575][15256:15258] CHIP:DMG: 		{
[1703485054.613622][15256:15258] CHIP:DMG: 			AttributePathIB =
[1703485054.613682][15256:15258] CHIP:DMG: 			{
[1703485054.613713][15256:15258] CHIP:DMG: 				Endpoint = 0x1,
[1703485054.613744][15256:15258] CHIP:DMG: 				Cluster = 0x201,
[1703485054.613776][15256:15258] CHIP:DMG: 				Attribute = 0x0000_0012,
[1703485054.613814][15256:15258] CHIP:DMG: 			}
[1703485054.613872][15256:15258] CHIP:DMG: 				
[1703485054.613912][15256:15258] CHIP:DMG: 			StatusIB =
[1703485054.613972][15256:15258] CHIP:DMG: 			{
[1703485054.614011][15256:15258] CHIP:DMG: 				status = 0x87 (CONSTRAINT_ERROR),
[1703485054.614067][15256:15258] CHIP:DMG: 			},
[1703485054.614106][15256:15258] CHIP:DMG: 			
[1703485054.614165][15256:15258] CHIP:DMG: 		},
[1703485054.614204][15256:15258] CHIP:DMG: 		
[1703485054.614262][15256:15258] CHIP:DMG: 	],
[1703485054.614302][15256:15258] CHIP:DMG: 	
[1703485054.614358][15256:15258] CHIP:DMG: 	InteractionModelRevision = 11
[1703485054.614393][15256:15258] CHIP:DMG: }
[1703485054.614443][15256:15258] CHIP:DMG: WriteClient moving to [AwaitingDe]
[1703485054.614498][15256:15258] CHIP:TOO: Response Failure: IM Error 0x00000587: General error: 0x87 (CONSTRAINT_ERROR)

jadhavrohit924, 2023-12-26:
> in heating mode（system-mode = 4), max_heat_setpoint_limit = 3500，max_heat_setpoint_limit = 2200,
write occupied-heating-setpoint = 2600, return error : 0x87 (CONSTRAINT_ERROR)

The value of system-mode does not disable the AUTO feature of the device, system-mode value only indicates current operating mode. Since your device have AUTO feature you are getting CONSTRAINT_ERROR

Kristian8606, 2023-12-30:
In order not to open a new problem, I will ask my question here.
On cooling else heating mode the callback function `app_driver_attribute_update` does not work when I try to set a target temperature I noticed that if I remove the function  `auto`  `app_driver_attribute_update` is worked.
Is this expected behavior?
This is the example I use.

```node::config_t node_config;
    node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);

    thermostat::config_t thermostat_config;
    endpoint_t *endpoint = thermostat::create(node, &thermostat_config, ENDPOINT_FLAG_NONE, NULL);
    cluster_t *thermostat_cluster = cluster::get(endpoint, Thermostat::Id);
   
   //heat
   cluster::thermostat::feature::heating::config_t heat_config;
   cluster::thermostat::feature::heating::add(thermostat_cluster, &heat_config);

    //cool 
   cluster::thermostat::feature::cooling::config_t cool_config;
   cluster::thermostat::feature::cooling::add(thermostat_cluster, &cool_config);    

    //auto
   cluster::thermostat::feature::auto_mode::config_t auto_config;
   cluster::thermostat::feature::auto_mode::add(thermostat_cluster, &auto_config); 

   //setback
   cluster::thermostat::feature::setback::config_t setback_config;
   cluster::thermostat::feature::setback::add(thermostat_cluster, &setback_config);```

jadhavrohit924, 2024-01-02:
@Kristian8606 Does the attribute updates in both the cases or you get CONSTRAINT_ERROR?

Kristian8606, 2024-01-02:
I don't get an error, I just don't get a callback when the `auto` function is enabled.

jadhavrohit924, 2024-01-08:
@Kristian8606 Can you please attach the complete logs to analyse?

dhrishi, 2024-01-18:
@Kristian8606 Can you please share the required logs, if you are still seeing this issue. If not, please close this.

dhrishi, 2024-01-29:
I am closing this now. Please re-open by providing the requested details.

closed 2024-01-29 by dhrishi