espressif/esp-matter #775: Room Air Conditioner Example Shows 'Not Supported' in Apple Home

mriksman, 2023-12-17:
**Describe the bug**
I downloaded the `room_air_conditioner` example and added it to Apple Home. However, it just shows up as 'Not Supported'.

iOS 17.2
HomePod 17.2
ESP32-Pico-D4-DevKit

**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id: `6b1f40b9b`
- SoC (eg: ESP32 or ESP32-C3): ESP32-Pico-D4
- Device Logs (Please attach the log file): Attached
- Host Machine OS: Ubuntu/Linux
- Commissioner app and versions if present:
- Commissioner's logs if present:

[AirConLog.txt](https://github.com/espressif/esp-matter/files/13694997/AirConLog.txt)
[image]

mriksman, 2023-12-20:
I just realised this example is for the new 'Room Air Conditioner' appliance in Matter 1.2 which hasn't been included in Apple yet.

closed 2023-12-20 by mriksman

anuarrozman, 2023-12-21:
@mriksman  Hii, Sorry but can you share where you get the information about your statement ? Thank you!!

mriksman, 2023-12-21:
@anuarrozman 
A quick Google has many articles talking about the release of Matter 1.2 on 23 October 2023
https://csa-iot.org/newsroom/matter-1-2-arrives-with-nine-new-device-types-improvements-across-the-board/

> Room Air Conditioners – While HVAC and thermostats were already part of Matter 1.0, standalone Room Air Conditioners with temperature and fan mode control are now supported.

I imagine Apple haven't updated their platform in such a short period of time; so for now you either must use a HVAC or Thermostat endpoint, or wait until Apple supports Matter 1.2.

anuarrozman, 2023-12-21:
Thanks @mriksman . I read the Matter 1.2 too but can't find any information about Apple haven't update their platform to support room air conditioner yet.