espressif/esp-matter #777: Adding an event for BLE deinit (DeviceEventType)

law-ko, 2023-12-18:
In current `esp_matter_core.cpp` there are several `DeviceEventType` events posted to `static void app_event_cb(const ChipDeviceEvent *event, intptr_t arg)`. Is there any chance to post events for BLE deinitialzed when the RAM is freed? 

Maybe [`kCHIPoBLEConnectionClosed`](https://github.com/project-chip/connectedhomeip/blob/7cbac9130d2b10d6bfdba873129dfbffa294cf8d/src/include/platform/CHIPDeviceEvent.h#L169C5-L169C30) or [`kCloseAllBleConnections`](https://github.com/project-chip/connectedhomeip/blob/7cbac9130d2b10d6bfdba873129dfbffa294cf8d/src/include/platform/CHIPDeviceEvent.h#L175C11-L175C11)

shubhamdp, 2023-12-18:
@law-ko `kCHIPoBLEConnectionClosed` and `kCloseAllBleConnections` are defined for different purpose and can not be used. We can add a different one though.
```
    /** Signals that BLE is deinitialized and heap memory is reclaimed */
    kBLEDeinitialized,
```

law-ko, 2023-12-18:
@shubhamdp Thank you for the response. Is this available with a patch or do we need to wait for your implementation?

shubhamdp, 2023-12-19:
@law-ko this is not available right now, but I'll upload patch here till this gets merged and updated on github

shubhamdp, 2023-12-19:
@law-ko please see if the attached patch works for you [0001-Added-an-event-when-BLE-is-deinitialized-and-memory-patch.txt](https://github.com/espressif/esp-matter/files/13714353/0001-Added-an-event-when-BLE-is-deinitialized-and-memory-patch.txt)

law-ko, 2023-12-20:
@shubhamdp I believe it is working. Thank you

closed 2023-12-21 by espressif-bot

shubhamdp, 2023-12-21:
@law-ko patch is merged in main branch