espressif/esp-matter #778: ThermostatRunningState support?

apexskier, 2023-12-18:
I want to indicate when a heater is "active", when the thermostat is actually telling it to heat when it's too cold in the room. Thermostat running state appears to be the appropriate attribute to set, but I see the following in the logs:

```
I (40592) chip[ZCL]: 0x3c12939cep 1 clus 0x0000_0201 attr 0x0000_0029 not supported
E (40602) esp_matter_attribute: Error updating Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000029 to matter: 0x86
```

I'm able to set some other attributes without issues

jadhavrohit924, 2023-12-27:
@apexskier Error 0x86 is UnsupportedAttribute. Did you add the attribute to the thermostat? Since ThermostatRunningState is optional attribute by default it will not be present on device, you have to add it yourself.

apexskier, 2023-12-28:
Thanks, that worked!

```cpp
esp_matter::cluster::thermostat::attribute::create_thermostat_running_state(thermostat_cluster, 0);
```

closed 2023-12-28 by apexskier