espressif/esp-matter #779: Can not commision ESP32C6 with thread

johnvigg, 2023-12-19:
**Describe the bug**
I followed the guide line by line and since I wanted to connect the esp32c6 with the lighting example using thread, I changed the 3 values using menuconfig. When I run the setup and try to pair, it gives and error. I have tried the same set up with wifi and it works, and I have tried setting up a thread example using a silicon labs chip, which also worked. 

**Environment**
- ESP-Matter Commit Id:f5a314a486e3f9b5cf71f48165cd44365de719cd (should be the latest one)
- ESP-IDF Commit Id: b1f40b9bfb91ec82fab4a60e5bfb4ca0c9b062f (latest as well) 
- SoC (eg: ESP32 or ESP32-C3): ESP32c6
- Device Logs (Please attach the log file):
[log.light.20231220013027.txt](https://github.com/espressif/esp-matter/files/13720547/log.light.20231220013027.txt)

- Host Machine OS: Ubuntu 22.04.3 LTS
- Commissioner app and versions if present: -
- Commissioner's logs if present: 
[screen.log](https://github.com/espressif/esp-matter/files/13720675/screen.log)
**Any additional details**

wqx6, 2023-12-25:
There is a BLE-15.4 co-exist issue for ESP32-C6 on the IDF commit you are using, and it should be fix in tag v5.1.2. Could you switch the IDF version to v5.1.2 and try again?

johnvigg, 2023-12-28:
> There is a BLE-15.4 co-exist issue for ESP32-C6 on the IDF commit you are using, and it should be fix in tag v5.1.2. Could you switch the IDF version to v5.1.2 and try again?

Hey, I changed to the version you mentioned, and I am now able to commission and everything works. Thank you for your help! I  will close this issue now.

closed 2023-12-28 by johnvigg