espressif/esp-matter #780: Too large DRAM .bss size occupation issue

YOGYUI, 2023-12-23:
**Describe the bug**
After updating CHIP sdk from commit id **181b0cb14ff007ec912f2ba6627e05dfb066c008**(tag v1.2.0.1) to **8e3d98c03904c14baeac495113eb58721797e838**, It is unable to flash application because of DRAM overflow issue.
```shell
/home/yogyui/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: matter-test-app.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/home/yogyui/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/home/yogyui/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/home/yogyui/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 3088 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```
I performed size analysis (using **idf.py size-files**)and found some interesting features like below.
(1) CHIP commit 181b0cb14ff007ec912f2ba6627e05dfb066c008 (previous version)
Object File                                 DRAM.bss
Server.cpp.obj                           19540
color-control-server.cpp.obj    1632
libCHIPDataModel.Interac        10752
level-control.cpp.obj                 640
attribute-storage.cpp.obj         6
on-off-server.cpp.obj               468
...

(2) CHIP commit 8e3d98c03904c14baeac495113eb58721797e838 (current version)
Object File                                 DRAM.bss
Server.cpp.obj                           19276
color-control-server.cpp.obj    **13188**
libCHIPDataModel.Interac        10752
level-control.cpp.obj                 **5252**
attribute-storage.cpp.obj         **4614**
on-off-server.cpp.obj               **2716**
...

I found that usage of static DRAM by CHIP related objects is enormously increased. 
ex) on-off-server object .bss usage increased by about 5.8 times

I did not dig CHIP source code yet...
Is there any way to decrease .bss usage of CHIP code just modifying sdkconfig or whatelse? (or should I wait for next commit?)

**Environment**
- ESP-Matter Commit Id: 30e5b4b91c9134abd056d05559660b252160802e
- ESP-IDF Commit Id: 6b1f40b9bfb91ec82fab4a60e5bfb4ca0c9b062f
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file):
- Host Machine OS: Ubuntu 22.04.3 LTS
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

elektro-NIK, 2023-12-23:
I have this issue too. Both: IDF v5.1.1 + ESP-Matter (30e5b4b91c9134abd056d05559660b252160802e) and docker image espressif/esp-matter:latest_idf_v5.1.1 (844f6bc7eea90a97f25923eeb44bc9e640ebe9733ce458a580b2fe96f7e85d87)

shubhamdp, 2023-12-27:
@YOGYUI @elektro-NIK Can you please share the application that you are trying, and its sdkconfig. I tried the [examples/light](https://github.com/espressif/esp-matter/tree/main/examples/light) from the SDK and below was for DRAM increase.

Older -> esp-matter: 5cd35a5 (connectedhomeip: [181b0cb14f](https://github.com/project-chip/connectedhomeip/tree/181b0cb14f))
Latest -> esp-matter: 397b967 (connectedhomeip : [8e3d98c039](https://github.com/project-chip/connectedhomeip/tree/8e3d98c039))

```
// esp32c3
libCHIP -> 128 bytes
libesp_matter -> 1132 bytes

// esp32
libCHIP -> 128 bytes
libesp_matter -> 592 bytes
```

elektro-NIK, 2023-12-27:
I tried to reproduce it on the clean environment, and it's visible on the bridge with multiple endpoints:
- [examples/zigbee_bridge](https://github.com/espressif/esp-matter/tree/main/examples/zigbee_bridge)
- changed `CONFIG_ESP_MATTER_MAX_DYNAMIC_ENDPOINT_COUNT` to 32
- esp32

| | Older (https://github.com/espressif/esp-matter/commit/72b0f263)    | Newer (https://github.com/espressif/esp-matter/commit/30e5b4b91c9134abd056d05559660b252160802e) |
| -- | -------- | ------- |
| libCHIP.a | 22863  | 22998    |
| libesp_matter.a | 29035 | 30823     |
| DRAM used | 97.8% | 99.4% |

Not so big numbers, like @YOGYUI, but my main project has up to 120 dynamic endpoints and it produces much bigger delta.

shubhamdp, 2023-12-28:
@elektro-NIK DRAM size would indeed depends on `CONFIG_ESP_MATTER_MAX_DYNAMIC_ENDPOINT_COUNT`, please check implementation of [color-control-cluster](https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/color-control-server/color-control-server.h#L255) and [level-control-cluster](https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/level-control/level-control.cpp#L101) in connectedhomeip repo and the struct size is > 20 bytes, so every dynamic cluster will add up to the DRAM increase.

@YOGYUI do you seeing the same problem?

YOGYUI, 2023-12-28:
@shubhamdp 
Sorry for the late response.
Please understand that I'am not possible to open my source code because of the company's policy.

Snippet of my sdkconfig is like below. (I'm developing bridge board for many other non-matter devices)
```
CONFIG_ESP_MATTER_MAX_DYNAMIC_ENDPOINT_COUNT=128
```

I just wonder why .bss usage was significantly increased just after updaing sdk(CHIP) **with same source code**...

shubhamdp, 2023-12-29:
@YOGYUI This 5X change that you mentioned in you original comment is not due to the change of submodule, but due to the change in value of config `CONFIG_ESP_MATTER_MAX_DYNAMIC_ENDPOINT_COUNT`.

With 16, below are the items in `.bss` for `on-off-server.ccp.obj`, probably the same increase for other object files.
```
 .bss._ZL14gEventControls                0x80
 .bss._ZN11OnOffServer8instanceE          0x8
 .bss._ZL11firstEffect                    0x4
 .bss._ZL18sOnOffSceneHandler           0x150
```

For 128,
```
 .bss._ZL14gEventControls               0x400
 .bss._ZN11OnOffServer8instanceE          0x8
 .bss._ZL11firstEffect                    0x4
 .bss._ZL18sOnOffSceneHandler           0x690 
```

At this point I would suggest you use the somewhat lower value for the dynamic endpoint count. Or there's the [Minimizing RAM Usage](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html) guide, check if you can reduce some dram/iram by setting some configurations from here.

YOGYUI, 2023-12-29:
@shubhamdp 
After modifying CONFIG_ESP_MATTER_MAX_DYNAMIC_ENDPOINT_COUNT as 100, problem had solved.

I'll close this issue.
But, I still wonder the reason why .bss usage had increased. 
I used "**same source code and same sdkconfig**" and only difference was CHIP commit.
I'll found out the reason with analyzing CHIP code diff on my own.
But, still thanks anyway~

closed 2023-12-29 by YOGYUI

shubhamdp, 2024-01-02:
@YOGYUI newer code will add new things and there will be minor increases in the `.bss` or `.text`. If you see, DRAM was overflowed by 3088 bytes, that what would be added by connectedhomeip upgrade.

I agree that this needs to be tackled and will certainly try to optimize the static as well as dynamic memory usage.