espressif/esp-matter #781: Apple Home cannot recognize the device

itatboy, 2023-12-28:
**Describe the bug**
I compiled the example firmware to esp32c2 using the following steps, but the device cannot be detected within Apple Home

cd esp-matter/examples/light_switch
idf.py set-target esp32c2
idf.py build
idf.py -p /dev/ttyUSB0 erase_flash
idf.py -p /dev/ttyUSB0 flash monitor

**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id:
- SoC (eg: ESP32 or ESP32-C3): ESP32-C2

- Device Logs (Please attach the log file):

[20:22:54.309]收←◆ESP-ROM:esp8684-api2-20220127
Build:Jan 27 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5c80,len:0x16bc
load:0x403acb70,len:0x974
load:0x403aeb70,len:0x2860
entry 0x403acb70
[0;32mI (21) boot: ESP-IDF v5.1.1 2nd stage bootloader[0m
[0;32mI (21) boot: compile time Dec 28 2023 12:09:40[0m
[0;32mI (21) boot: chip revision: v1.0[0m
[0;32mI (24) boot.esp32c2: MMU Page Size  : 64K[0m
[0;32mI (29) boot.esp32c2: SPI Speed      : 60MHz[0m
[0;32mI (33) boot.esp32c2: SPI Mode       : DIO[0m
[0;32mI (38) boot.esp32c2: SPI Flash Size : 4MB[0m
[0;32mI (43) boot: Enabling RNG early entropy source...[0m
[0;32mI (48) boot: Partition Table:[0m
[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (59) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000[0m
[0;32mI (67) boot:  1 nvs              WiFi data        01 02 00010000 0000c000[0m
[0;32mI (74) boot:  2 nvs_keys         NVS keys         01 04 0001c000 00001000[0m
[0;32mI (81) boot:  3 otadata          OTA data         01 00 0001d000 00002000[0m
[0;32mI (89) boot:  4 phy_init         RF data          01 01 0001f000 00001000[0m
[0;32mI (96) boot:  5 ota_0            OTA app          00 10 00020000 001e0000[0m
[0;32mI (104) boot:  6 ota_1            OTA app          00 11 00200000 001e0000[0m
[0;32mI (111) boot:  7 fctry            WiFi data        01 02 003e0000 00006000[0m
[0;32mI (119) boot: End of partition table[0m
[0;32mI (123) esp_image: segment 0: paddr=00020020 vaddr=3c0e0020 size=333b8h (209848) map[0m

[20:22:54.587]收←◆[0;32mI (180) esp_image: segment 1: paddr=000533e0 vaddr=3fca74e0 size=02084h (  8324) load[0m
[0;32mI (183) esp_image: segment 2: paddr=0005546c vaddr=40380000 size=074d4h ( 29908) load[0m
[0;32mI (194) esp_image: segment 3: paddr=0005c948 vaddr=00000000 size=036d0h ( 14032) [0m
[0;32mI (197) esp_image: segment 4: paddr=00060020 vaddr=42000020 size=de6c4h (911044) map[0m

[20:22:54.947]收←◆[0;32mI (414) boot: Loaded app from partition at offset 0x20000[0m
[0;32mI (415) boot: Disabling RNG early entropy source...[0m
[0;32mI (426) cpu_start: Unicore app[0m
[0;32mI (427) cpu_start: Pro cpu up.[0m
[0;32mI (435) cpu_start: Pro cpu start user code[0m
[0;32mI (435) cpu_start: cpu freq: 120000000 Hz[0m
[0;32mI (435) cpu_start: Application information:[0m
[0;32mI (438) cpu_start: Project name:     light_switch[0m
[0;32mI (444) cpu_start: App version:      1.0[0m
[0;32mI (448) cpu_start: Compile time:     Dec 28 2023 12:09:32[0m
[0;32mI (454) cpu_start: ELF file SHA256:  b1a9a38f9490e373...[0m
[0;32mI (460) cpu_start: ESP-IDF:          v5.1.1[0m
[0;32mI (465) cpu_start: Min chip rev:     v1.0[0m
[0;32mI (470) cpu_start: Max chip rev:     v2.99 [0m
[0;32mI (475) cpu_start: Chip rev:         v1.0[0m
[0;32mI (480) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (487) heap_init: At 3FCB75E0 len 00025590 (149 KiB): D/IRAM[0m
[0;32mI (493) heap_init: At 3FCDCB70 len 0000294C (10 KiB): STACK/DRAM[0m
[0;32mI (501) spi_flash: detected chip: gd[0m
[... 119 lines cut ...]
chan:13,max_power:80
chan:14,max_power:80
W (1627) wifi:Haven't to connect to a suitable AP now!
[0;32mI (1632) chip[DL]: Done driving station state, nothing else to do...[0m
[0;32mI (1640) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)[0m
[0;32mI (1647) chip[DL]: Device already advertising, stop active advertisement and restart[0m
[0;32mI (1656) NimBLE: GAP procedure initiated: stop advertising.
[0m
[0;32mI (1663) NimBLE: GAP procedure initiated: advertise; [0m
[0;32mI (1668) NimBLE: disc_mode=2[0m
[0;32mI (1672) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40[0m
[0;32mI (1682) NimBLE: 
[0m
[0;32mI (1685) app_main: Commissioning window opened[0m
[0;32mI (1691) main_task: Returned from app_main()[0m

- Host Machine OS:Ubuntu20.04 server
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**

[image]

...

itatboy, 2023-12-28:
In addition, there is a strange phenomenon: when executing idf.py - p/dev/ttyUSB0 flash monitor, the monitored startup log is garbled. The serial port assistant is used, and the device baud rate is set to 76800 to obtain the above device logs normally.

I hope you can provide some ideas to solve the problem. Thank you very much

Sdkconfig is attached

[image]
[sdkconfig.log](https://github.com/espressif/esp-matter/files/13787017/sdkconfig.log)

shubhamdp, 2023-12-29:
@itatboy light_switch won't appear in the Apple Home as it contains client clusters. The devices containing the server clusters will appear in there. Try generic_switch or light if you want the device be controllable from Home app.

itatboy, 2023-12-29:
@shubhamdp 
Thank you for your reply. I am using the same instance project light_ Switch, only make the following changes:
Idf.py set target esp32c3
The remaining steps remain unchanged, and the compilation is successfully burned to the ESP32C3 development board. Apple Home can detect the device normally and correctly add it to Home.
I am puzzled why this situation occurs. Is it not compatible with ESP32C2? I am using the ESP8684 devkitc-02 development board

itatboy, 2023-12-29:
Good, according to your prompt, Apple Home is now able to add accessories smoothly. Thank you. My understanding of esp matter is still far from sufficient, and this should not have been a problem. Can you help me close this issue? Thank you

closed 2023-12-30 by dhrishi