espressif/esp-matter #782: Regarding transferring strings using matter protocol

johnvigg, 2023-12-29:
**Describe the question/query that you have**
I am looking for a way to transfer strings to an esp32c6 which will be displayed on an lcd using the matter protocol.

**Additional context**

Currently I have a rpi4 running CHIP and I am able to commission the esp32c6 and run the light example.

Now I want to be able to use a specific cluster to transfer a string that I will pass as an argument using the CHIP controller.

I was not able to find a cluster that uses strings, probably since this protocol wasn't intended to be used that way, but is there maybe a roundabout way of doing this?
Maybe there a simpler way to do this, or do I have to create a custom cluster?

Any advice is appreciated.

shripad621git, 2024-01-02:
@johnvigg , you can modify the attribute_update_cb in light example as follows:
```
static esp_err_t app_attribute_update_cb(attribute::callback_type_t type, uint16_t endpoint_id, uint32_t cluster_id,
                                         uint32_t attribute_id, esp_matter_attr_val_t *val, void *priv_data)
{
    esp_err_t err = ESP_OK;

    if (type == PRE_UPDATE) {
        /* Driver update */
        app_driver_handle_t driver_handle = (app_driver_handle_t)priv_data;
        err = app_driver_attribute_update(driver_handle, endpoint_id, cluster_id, attribute_id, val);
        if (endpoint_id == 0 && cluster_id == BasicInformation::Id && attribute_id == BasicInformation::Attributes::NodeLabel::Id)
        {
            char * label = (char*)malloc(val->val.a.s);
            memcpy(label, val->val.a.b, val->val.a.s);
            label[val->val.a.s] = '\0';
            printf("Label %s",label);
        }
        
    }

    return err;
}

```
Instead of using the printf ,you can write the logic to display the string on the lcd.
You need to set the node-label using chip-tool using the command given below:
```
chip-tool basicinformation write node-label "light101" <node-id> 0

```

johnvigg, 2024-01-02:
@shripad621git , I followed what you recommended and it works perfectly. Thank you very much for your help! I will close issue now.

closed 2024-01-02 by johnvigg