espressif/esp-matter #783: blemesh example doesn't build

jonsmirl, 2023-12-30:
If I am not using blemesh but I am using ble, do I still need the custom platform?

~~~
meip/connectedhomeip/config/esp32/third_party/connectedhomeip/third_party/nlio/repo/include -c ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/asn1/ASN1OID.cpp -o obj/third_party/connectedhomeip/src/lib/asn1/libASN1.ASN1OID.cpp.o
In file included from ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/ble/BleConfig.h:40,
                 from ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/core/CHIPConfig.h:42,
                 from ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/asn1/ASN1Config.h:29,
                 from ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/asn1/ASN1Error.h:32,
                 from ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/asn1/ASN1.h:33,
                 from ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/asn1/ASN1OID.cpp:37:
gen/include/platform/CHIPDeviceBuildConfig.h:22:38: fatal error: platform/ESP32_custom/CHIPPlatformConfig.h: No such file or directory
   22 | #define CHIP_PLATFORM_CONFIG_INCLUDE <platform/ESP32_custom/CHIPPlatformConfig.h>
      |                                      ^
compilation terminated.
[90/375] c++ obj/third_party/connectedhomeip/src/app/libCHIPDataModel.CommandHandler.cpp.oninja: build stopped: subcommand failed.
[545/1261] Building C object esp-idf/wpa_supplicant/CMakeFiles/__idf_wpa_supplicant.dir/src/rsn_supp/wpa.c.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/jonsmirl/aosp/esp-matter/examples/blemesh_bridge/build/log/idf_py_stderr_output_1172581 and /home/jonsmirl/aosp/esp-matter/examples/blemesh_bridge/build/log/idf_py_stdout_output_1172581
jonsmirl@ares:~/aosp/esp-matter/examples/blemesh_bridge$ 
~~~

shubhamdp, 2024-01-02:
@jonsmirl This was broke after a submodule update and will be fixed with https://github.com/project-chip/connectedhomeip/pull/31188.

closed 2024-01-07 by jonsmirl

shubhamdp, 2024-01-08:
@jonsmirl The fix is merged into esp-matter main branch