espressif/esp-matter #784: ESP32 Matter Door Lock Example in connectedhomeip

bilalmalik76, 2023-12-31:
Is it possible to modify the ESP32 Matter Door Lock Example, available in connectedhomeip, to accommodate up to 10 door lock endpoints? If so, what is the recommended approach for implementing this?

We have a specific requirement wherein we intend to utilize the ESP32 Matter Door Lock Example as a hub for an additional 10 door locks. The communication between the hub and the other locks will be facilitated over RF. The Matter door lock hub is expected to interact with the Matter app on behalf of these 10 door locks. What would be the best strategy to proceed with this customization?

jadhavrohit924, 2024-01-02:
@bilalmalik76 It is possible to have 10 door lock endpoints in the examples available in connectedhomeip, to have to endpoints you will have to edit the .zap of the door-lock app using the zap-tool available in `scripts/tools/zap/run_zaptool.sh`.

I would like to suggest a briged, an examples in esp-matter for bridge, the bridge will communicate with the matter app on behalf of the door locks. Just that we don't have any which communicates using RF so you will have to build that one.

bilalmalik76, 2024-01-02:
@jadhavrohit924 Thansk I wil try to modify .zap file and one more thing with matter bridge, all 10 devices will appear on any ecosystem of an app as door locks?

bilalmalik76, 2024-01-02:
@jadhavrohit924 which door-lock.zap file I need to mofiy and how to generate door-lock.zap file with 10 end points?

jadhavrohit924, 2024-01-02:
> @jadhavrohit924 which door-lock.zap file I need to mofiy and how to generate door-lock.zap file with 10 end points?

Yes use the script `scripts/tools/zap/run_zaptool.sh` run
`scripts/tools/zap/run_zaptool.sh examples/lock-app/lock-common/lock-app.zap` and add the endpoints that you need.

> @jadhavrohit924 Thansk I wil try to modify .zap file and one more thing with matter bridge, all 10 devices will appear on any ecosystem of an app as door locks?

Yes

bilalmalik76, 2024-01-02:
@jadhavrohit924 How can I add endpoints to lock-app.zap file, it looks like a generated file with some tool.

bilalmalik76, 2024-01-02:
@jadhavrohit924 is this a right way to add end points:

`  {
      "endpointTypeName": "MA-doorlock",
      "endpointTypeIndex": 1,
      "profileId": 259,
      "endpointId": 1,
      "networkId": 0
    }
    {
      "endpointTypeName": "MA-doorlock",
      "endpointTypeIndex": 1,
      "profileId": 259,
      "endpointId": 2,
      "networkId": 0
    }
    {
      "endpointTypeName": "MA-doorlock",
      "endpointTypeIndex": 1,
      "profileId": 259,
      "endpointId": 3,
      "networkId": 0
    }
    {
      "endpointTypeName": "MA-doorlock",
      "endpointTypeIndex": 1,
      "profileId": 259,
      "endpointId": 4,
      "networkId": 0
    }
    {
      "endpointTypeName": "MA-doorlock",
      "endpointTypeIndex": 1,
      "profileId": 259,
      "endpointId": 5,
      "networkId": 0
    }
    {
      "endpointTypeName": "MA-doorlock",
      "endpointTypeIndex": 1,
      "profileId": 259,
      "endpointId": 6,
      "networkId": 0
    }
    {
      "endpointTypeName": "MA-doorlock",
      "endpointTypeIndex": 1,
      "profileId": 259,
      "endpointId": 7,
      "networkId": 0
    }
    {
      "endpointTypeName": "MA-doorlock",
      "endpointTypeIndex": 1,
      "profileId": 259,
      "endpointId": 8,
      "networkId": 0
    }
    {
      "endpointTypeName": "MA-doorlock",
      "endpointTypeIndex": 1,
      "profileId": 259,
      "endpointId": 9,
      "networkId": 0
    }
    {
      "endpointTypeName": "MA-doorlock",
      "endpointTypeIndex": 1,
      "profileId": 259,
      "endpointId": 10,
      "networkId": 0
    }`

bilalmalik76, 2024-01-02:
@jadhavrohit924 I am getting this error:
[image]

bilalmalik76, 2024-01-02:
@jadhavrohit924 I have added end points and build flash code to ESP32. But only one door lock is showing on home app after commissioing. Do I need to do anything else after adding end points?

bilalmalik76, 2024-01-08:
@jadhavrohit924 if we use bridge-app, how much effort it will require to add door lock related functionality into it? We have not found any non-zap based esp matter door lock example

jadhavrohit924, 2024-01-10:
@bilalmalik76 esp-matter is the non-zap based data model, you just have to create appropriate device type using single API I would suggest please take a look how devices are being created in [light app](https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L169) of esp-matter in same way you can create a door lock. Here are the [all available device types](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_endpoint.h)

bilalmalik76, 2024-01-11:
@jadhavrohit924 ending simply endpoint does not work for the door lock. I have already tried it.

jadhavrohit924, 2024-01-11:
What is the issue that you faced?

jadhavrohit924, 2024-01-16:
@bilalmalik76 Solution to this issue has been discussed [here](https://github.com/espressif/esp-matter/issues/790#issuecomment-1882880161), please check

bilalmalik76, 2024-01-16:
@jadhavrohit924 and that solution does not work at all.

jadhavrohit924, 2024-01-16:
Can you please share the patch that you have added? I'll take a look and check what is missing.

dhrishi, 2024-01-31:
@bilalmalik76 With the same setup as mentioned [here](https://github.com/espressif/esp-matter/issues/809#issuecomment-1917275696), I am able to commission the door lock to Apple Home and then control the same using the lock and unlock commands

dhrishi, 2024-03-12:
Closing this issue now, as per the comment above.

closed 2024-03-12 by dhrishi