espressif/esp-matter #785: "Fabric Name" chages from "Matter Test" to "Chai" when Info Provider is changed from Test to Fabric

a-tobias, 2024-01-01:
Setup : 1x ESP32 Controller and 1x Thread Sleepy Device (ESP32).
The Thread Node is added to Apple Homekit. If I add the Thread Node to the ESP32 Controller, Homekit informs me that I added the node to another Fabric called "Matter Test".
If i change the Device Info Provider and attestation to Fabric and  do the same as described above, Homekit tells me the ESP32 Controller Fabric is called "Chai".

Where does this name come from ? How can I changed it ?

wqx6, 2024-01-02:
What device info provider and attestation do you use when the Fabric is "Chai"? Are they on the end-device or the controller?

wqx6, 2024-01-02:
Normally the HomeKit will use the FabricLabel as the Fabric name, so you can change the fabric name by UpdateFabricLabel command in OperationalCredentials cluster.

wqx6, 2024-01-02:
The HomeKit will use the vendor id from the DeviceInstanceInfoProvider to get the Fabric name from DCL, so when you change the DeviceInstanceInfoProvider, the Fabric name will also be changed.

a-tobias, 2024-01-02:
Thanks for the quick reply...
Regarding the first question - yes i forgot to mention that I changed the Data Provider of the Controller only.

Basically, I have already assumed that this has something to do with the data within the factory partition but when i use the mfg_tool i used the following to create the binary : 

-cn "ESP32 Device" -v 0xFFF2 -p 0x8001 --hw-ver 0x01 --pai \
--vendor-name  <some string> \
--hw-ver-str "v0.1" \
--product-name <some string> \

So I`am still wondering where "Chai" comes from.

wqx6, 2024-01-03:
The HomePod cannot find vendor id 0xFFF2 in [DCL MainNet](https://dcl.iotledger.io/vendors) so It will use "Chai" as the vendor name, and for vendor ID 0xFFF1(Default value) HomePod will use "Matter Test". If you use 0x131B as the controller's vendor ID it will be "Espressif System" since HomePod can find it in DCL.

a-tobias, 2024-01-05:
Ok. Thanks for clarification. I close this issue.

closed 2024-01-05 by a-tobias