espressif/esp-matter #786: ESP32c3 as matter Hub

hetal-mpc, 2024-01-02:
**Describe the question/query that you have**
A clear and concise description of what you want to ask.

**Additional context**
...
I am trying matter examples  on esp32c3 and everytime i flash the example on device and see a pop-up appears on android phone for new matter device found but when I try to add that device I get error of matter hub is required. 
Can you please suggest how I can implement esp32c3 as matter hub?

dhrishi, 2024-01-02:
@hetal-mpc 
> a pop-up appears on android phone for new matter device found but when I try to add that device I get error of matter hub is required.

The ecosystems require a hub for commissioning Matter devices. In this case, the hub is a Google device like a NestHub or a Google Home mini etc. Supported hubs list [here](https://developers.home.google.com/matter/eap#hubs)

You can use the host-based chiptool to commission Matter devices, by following the setup instructions https://docs.espressif.com/projects/esp-matter/en/latest/esp32c3/developing.html#development-setup . This does not require an additional hub

hetal-mpc, 2024-01-08:
Thanks @dhrishi it helps
Bu in the light_switch example it worked upto commissioning, after that if when I send  `onoff toggle 0x7283 0x1`  command using chip-tool is doesn't show any action on that. it shows UNSUPPORTED_CLUSTER but where i can find which clusters are supported with light_switch example.

dhrishi, 2024-01-08:
@hetal-mpc The light_switch example has on_off client cluster. In short, it can be used to control on_off server cluster devices (like a light) over Matter. You can instead use the light example and test

Please refer the [Matter specifications](https://github.com/espressif/esp-matter?tab=readme-ov-file#matter-specifications) for additional details.

closed 2024-01-08 by dhrishi