espressif/esp-matter #787: Fan Control Cluster::PercentSetting attribute doesn't work in iOS Home App

adailydose, 2024-01-03:
**Describe the question/query that you have**
Does anyone know if fan_control.percent_setting is supported in the iOS Home app? Or have any suggestions on how to find out?

I setup a ESP32 with fan control, and I can see and adjust the fan percentage in the Google Home app, but only see ON/OFF control in the iOS Home app.

**Additional context**
I raised the question in the Apple Developer forum as well.

shripad621git, 2024-01-03:
@adailydose , Please specify the following details:
1.esp-matter commit id
2.esp-idf version used
3.iOS Home app version

shripad621git, 2024-01-03:
@adailydose, I have tried the `fan control` on` esp-matter latest main branch.`
`esp-matter Commit id:` 2f692ac10a09ed27c223499ecdc458f7730d544d
Application Used: all-clusters-app  
Command to create fan device : `create --device_type=fan` (in console)
`HomePod Version:` 17.2 
`iOS version`: 
The `PercentSetting` attribute worked fine on the homepod.
The screenshots are attached below:
[image]
[image]

adailydose, 2024-01-04:
Hi @shripad621git, looks like this works after I update and restart my Apple TV to the latest version. Thank you!

closed 2024-01-04 by adailydose