espressif/esp-matter #788: mfg_tools requirements.txt is missing ecdsa module

gregbreen, 2024-01-03:
**Describe the bug**
If you install the mfg_tool requirements.txt in a virtual environment, you will suffer the following error when you try to run the tool.

```
Traceback (most recent call last):
  File "/home/ubuntu/environment/esp-matter/tools/mfg_tool/./mfg_tool.py", line 49, in <module>
    from spake2p import generate_verifier
  File "/home/ubuntu/environment/esp-matter/connectedhomeip/connectedhomeip/scripts/tools/spake2p/spake2p.py", line 25, in <module>
    from ecdsa.curves import NIST256p
ModuleNotFoundError: No module named 'ecdsa'
```

**Environment**
- ESP-Matter Commit Id: a7c1d96
- ESP-IDF Commit Id: b6a66b7

shubhamdp, 2024-01-03:
@gregbreen, thanks for raising the issue, initially, mfg-tool wasn't intended for standalone installation, so the `ecdsa `library is installed from [this source](https://github.com/project-chip/connectedhomeip/blob/master/scripts/setup/requirements.esp32.txt#L10).

We are currently in the process of releasing the mfg-tool to PyPi. Once completed, the mfg-tool can be installed using pip, and the ecdsa dependency will be handled seamlessly

gregbreen, 2024-01-03:
@shubhamdp thanks for the explanation. Please close this issue if it's not helpful.

shubhamdp, 2024-02-16:
@gregbreen We have pushed the mfg-tool here on pypi: https://pypi.org/project/esp-matter-mfg-tool/1.0.0 with all the resolved dependencies. Please give it a try.

closed 2024-02-16 by dhrishi