espressif/esp-matter #789: Unable to find the configration code with matter

Whiteliondnd8, 2024-01-05:
**Describe the question/query that you have**
I am encountering an issue with testing my project developed using the ESP-Matter SDK in conjunction with Alexa. Specifically, I am unable to find any configuration code or barcode that is typically required for integrating and testing with Alexa. Could someone guide me on how to properly set up and test my ESP-Matter SDK project with Alexa, or provide insights on where to find the necessary configuration details and barcode?

**Additional context**

...

jonsmirl, 2024-01-07:
It is in the manual
https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning-and-control

Whiteliondnd8, 2024-01-08:
If i costomized matter light example code then how can i test

shubhamdp, 2024-01-08:
@Whiteliondnd8 please take a look at [production consideration guide](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/production.html).

dhrishi, 2024-01-15:
Closing this now. Ideally, you should be able to create a custom QR code using the documentation

closed 2024-01-15 by dhrishi