espressif/esp-matter #790: ESP Matter Non-ZAP Door Lock Example

bilalmalik76, 2024-01-06:
Do you have any ESP Matter Non-ZAP based Door Lock Example?

shubhamdp, 2024-01-08:
@bilalmalik76 you can take a look at https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#developing-your-product.

You can also take a look at [all-device-type](https://github.com/espressif/esp-matter/tree/main/examples/all_device_types_app) example. This has the door-lock.

bilalmalik76, 2024-01-08:
@shubhamdp I already tried this example and able to commission with Apple home app but get this error when send lock and unlock command from home app and state does not change at all:
[image]

VaishaliAvhale, 2024-01-09:
@bilalmalik76 We are looking into the problem and will drop an update here soon.

shubhamdp, 2024-01-09:
@bilalmalik76 Please take a look at https://github.com/espressif/esp-matter/issues/176

You will need to implement few APIs in order to get the door lock working and that is discussed in that issue.

dhrishi, 2024-01-15:
Closing as the resolution is provided

closed 2024-01-15 by dhrishi