espressif/esp-matter #791: Line : 265 - C:/Espressif/frameworks/esp-idf-v5.1.2/components/log/include/esp_log.h:265:27: error: format '%x' expects argument of type 'unsigned int', but argument 6 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]

debasish30, 2024-01-10:
**Describe the bug**
ESP-IDF git package build compiler issue.

**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id:
- SoC (eg: ESP32 or ESP32-C3): 
[idf_py_stdout_output_22312.zip](https://github.com/espressif/esp-matter/files/13888621/idf_py_stdout_output_22312.zip)

- Device Logs (Please attach the log file):
- Host Machine OS:
- Commissioner app and versions if present:
- Commissioner's logs if present:

**Any additional details**
...

dhrishi, 2024-01-11:
@debasish30 Which example are you trying to build? Please provide the ESP Matter, ESP IDF commit id and other details.

debasish30, 2024-01-11:
@dhrishi Thank you for your immediate response.

To be frank i could not find the info on ESP-Matter Commit Id and ESP-IDF Commit Id.
Does it mean ESP-IDF tool version ? if yes then its 5.1.2

dhrishi, 2024-01-11:
@debasish30 Which example are you trying out? The command to get the head commit:
`git rev-parse HEAD` 

Can you please provide step wise details about your OS, setup, and installation.

dhrishi, 2024-01-12:
@debasish30 I am closing this issue. Please provide the necessary details and I will re-open

closed 2024-01-12 by dhrishi