espressif/esp-matter #792: Add static assert for nimble ble host flow control and chip over ble menu config options

damian-kurek-wizzdev, 2024-01-10:
I run into the same issue as here [](https://github.com/project-chip/connectedhomeip/issues/6839).
Adding static assert to code that nimble BLE host flow control and Chip over BLE can't be enabled at the same time would save a lot of time for me and others in future

shripad621git, 2024-01-12:
@damian-kurek-wizzdev , I have tried out the light example from esp-matter latest main branch using the two options: i)`CONFIG_BT_NIMBLE_HS_FLOW_CTRL=y ` which is  nimble BLE host flow control enabled 
ii)`CONFIG_ENABLE_CHIPOBLE=y `i.e .Chip over BLE enabled on` esp32 `and `esp32c3` respectively.
My commissioning was successful without any errors regarding ble connection.
Can you please specify the `esp-matter commit id` ,`esp-idf version `and the `SoC` you are using?

damian-kurek-wizzdev, 2024-01-12:
@shripad621git 
esp-idf 5.1.1 -> I'm updating my project to 5.1.2 can check it there
esp-matter main branch from 2 weeks ago 
esp32c6

shripad621git, 2024-01-12:
@damian-kurek-wizzdev , while we debug the reason of  failure with `esp32c6`, I would like to understand that do you have any specific use-case requirement with respect to both options being enabled in case of` esp32c6 `as the option `CONFIG_BT_NIMBLE_HS_FLOW_CTRL` is disabled by default for other SoCs other than esp32. You can check it out [here](https://github.com/espressif/esp-idf/blob/master/components/bt/host/nimble/Kconfig.in#L342C2-L342C2).

damian-kurek-wizzdev, 2024-01-12:
@shripad621git I think that CONFIG_BT_NIMBLE_HS_FLOW_CTRL was checked by someone in my team during development and we don't need this option. It seems that my code runs fine without it.
I noticed an issue and concluded that matter and CONFIG_BT_NIMBLE_HS_FLOW_CTRL are not compatible. 
 I created an issue so other developers can have an easier time. 
Thanks for asking.

dhrishi, 2024-01-31:
The flow control mechanism in the controller should be used by default

closed 2024-01-31 by dhrishi