espressif/esp-matter #793: tlsf_free_heap_tlsf assert triggered after ble disconnection

damian-kurek-wizzdev, 2024-01-11:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment**
- ESP-Matter Commit Id: main
- ESP-IDF Commit Id: 5.1.1
- SoC (eg: ESP32 or ESP32-C3): c6
- Device Logs (Please attach the log file):
- I (245986) chip[BLE]: Releasing end point's BLE connection back to application.
I (245986) esp_matter_core: BLE Disconnected

assert failed: tlsf_free heap_tlsf.c:907 (!block_is_free(block) && "block already marked as free")
Stack dump detected
Core  0 register dump:
MEPC    : 0x408005e0  RA      : 0x4080917a  SP      : 0x4084c880  GP      : 0x408143c0  
0x408005e0: panic_abort at /home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/esp_system/panic.c:452

0x4080917a: __ubsan_include at /home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x407f4760  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000001  S1      : 0x4084ca02  A0      : 0x4084c8d8  A1      : 0x40817859  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x4082a000  
A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x00000065  S3      : 0x4084c8d8  
S4      : 0x4084c8d8  S5      : 0x4004d150  S6      : 0x40825870  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

panic_abort (details=details@entry=0x4084c8d8 "assert failed: tlsf_free heap_tlsf.c:907 (!block_is_free(block) && \"block already marked as free\")") at /home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/esp_system/panic.c:452
452	    *((volatile int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
#0  panic_abort (details=details@entry=0x4084c8d8 "assert failed: tlsf_free heap_tlsf.c:907 (!block_is_free(block) && \"block already marked as free\")") at /home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/esp_system/panic.c:452
#1  0x4080917a in esp_system_abort (details=details@entry=0x4084c8d8 "assert failed: tlsf_free heap_tlsf.c:907 (!block_is_free(block) && \"block already marked as free\")") at /home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/esp_system/port/esp_system_chip.c:84
#2  0x4080d178 in __assert_func (file=<optimized out>, line=<optimized out>, func=<optimized out>, expr=<optimized out>) at /home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/newlib/assert.c:81
#3  0x40028392 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
ELF file SHA256: 801b242ae1917999

- Host Machine OS: Ubuntu 22.04
- Commissioner app and versions if present: Google home
- Commissioner's logs if present:

**Any additional details**
After running commissioning via BLE esp32 crashes with assert failed

shubhamdp, 2024-01-11:
This crash is fixed in [v5.1.2](https://github.com/espressif/esp-idf/tree/v5.1.2), Please use the newer esp-idf release.

damian-kurek-wizzdev, 2024-01-11:
Oh no. I will update it and check.

dhrishi, 2024-01-12:
@damian-kurek-wizzdev Please let us know if it fixes the issue. ESP Matter will soon be officially updated to use IDF v5.1.2

damian-kurek-wizzdev, 2024-01-17:
@dhrishi  @shubhamdp After the update it works.
 I got this error when nimble host control option was checked (I don't need it).
 (47838) chip[DL]: CHIPoBLE advertising stopped
W (48038) App: Device is in operation mode off - not heating
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
Stack dump detected
MEPC    : 0x42071eb2  RA      : 0x4206c2da  SP      : 0x40858ba0  GP      : 0x40817440  
0x42071eb2: ble_hs_flow_track_data_mbuf at /home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_flow.c:217

0x4206c2da: ble_hs_rx_data at /home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:700

TP      : 0x4080091c  T0      : 0x40022494  T1      : 0x4206c2ce  T2      : 0x02c176d6  
0x4080091c: heap_caps_alloc_failed at /home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/heap/heap_caps.c:98

0x40022494: multi_heap_internal_unlock in ROM

0x4206c2ce: ble_hs_rx_data at /home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:692

S0/FP   : 0x4082fa0c  S1      : 0x408542d0  A0      : 0x4082fa0c  A1      : 0x00000000  
A2      : 0x4082fa0c  A3      : 0x00000100  A4      : 0x00000001  A5      : 0x4103cddc  
A6      : 0x0000000c  A7      : 0x00000000  S2      : 0x4082fa0c  S3      : 0x4082c000  
S4      : 0x00000002  S5      : 0x00000000  S6      : 0x408543cc  S7      : 0x0000000b  
S8      : 0x421af000  S9      : 0x00000896  S10     : 0x000008aa  S11     : 0x4084bc18  
T3      : 0x00000006  T4      : 0x00000000  T5      : 0x02c18a0b  T6      : 0x02c17626  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x4103cddc  
0x40800001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

0x42071eb2 in ble_hs_flow_track_data_mbuf (om=om@entry=0x4082fa0c) at /home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_flow.c:217
217	    ble_hs_flow_mbuf_conn_handle[idx] = BLE_HCI_DATA_HANDLE(hdr->hdh_handle_pb_bc);
#0  0x42071eb2 in ble_hs_flow_track_data_mbuf (om=om@entry=0x4082fa0c) at /home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_flow.c:217
#1  0x4206c2da in ble_hs_rx_data (om=0x4082fa0c, arg=<optimized out>) at /home/dkurek/Projects/mill/mill-esp32/main/externals/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:698
#2  0x42084f70 in r_ble_ll_conn_rx_data_pdu ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
ELF file SHA256: 3396a2964a2c98a2

closed 2024-01-17 by damian-kurek-wizzdev