espressif/esp-matter #798: Extract active dataset using matter console

Xploder, 2024-01-15:
Since we can't get ot-cli running with the matter console simultaneously, is there a way to show the active thread dataset of a commissioned device using the matter console?

I'm trying to extract the dataset of an Amazon Thread network using an ESP32-C6 and the light example configures via c6_thread sdkconfig.

Xploder, 2024-01-16:
If it doesn't work through the matter console, It'd be enough for me to let it print the thread dataset to the serial log, if someone could point me in the right direction to add it to the light example it'd be much appreciated.

I've also dumped flash and see the network name in the nvs partition section of the dump but can't make sense of the rest.

shubhamdp, 2024-01-16:
@Xploder you can print the dataset during commissioning, please check [`req.operationalDataset`](https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/network-commissioning/network-commissioning.cpp#L616C41-L616C82), its a ByteSpan, you can use the `.data()` and `.size()` methods on it to get the actual data and size of that span and print it.

Xploder, 2024-01-16:
Thanks! 
Unfortunately I'm not a programmer but I'll try to modify network-commissioning.cpp and use "ChipLogProgress" to print req.operationalDataset.data(). Not sure how I would use the span though, if you could lead me to the right direction it'd be glad.

It'll be great to have a little "ESP32 Thread Credential Extractor" device!

Xploder, 2024-01-16:
You're so helpful, thanks! Couldn't have figured this out myself.
I will test it today after work.

Xploder, 2024-01-16:
It worked beautifully and I was able to extract the TLV dataset of my Amazon Echo Border Router. 
Again - thank you very much for your help!

closed 2024-01-16 by Xploder