espressif/esp-matter #799: Change Firmware Version shown (e.g. in HomeKit)

a-tobias, 2024-01-16:
I would like to change the firmware version provided by a matter node after a software update.
I already tried the following :
`    // Basic Information
    static char *SW_VER = "1.1.6";
    static char *SERIAL_NO = "1234";

    endpoint_t *endpoint_0 = endpoint::get(node, 0);
    cluster_t *inf_cluster = cluster::get(endpoint_0, BasicInformation::Id);
    cluster::basic_information::attribute::create_software_version_string(inf_cluster, SW_VER, (uint16_t)(sizeof(SW_VER)));
    cluster::basic_information::attribute::create_serial_number(inf_cluster, SERIAL_NO, (uint16_t)(sizeof(SERIAL_NO)));`

In this case I get a new parameter shown in Homekit (Serialnumber) but this is only a long hex-string and not the serialnumber i defined in an array of char (not that important).

After some research i guess that my code above is not the way it should be done since it already exists in endpoint 0 ?!
After some tests the firmware string has changed but not to a value i defined. Furthermore i cannot change it anymore even if i flash the thread node again (after erase_flash).

So - is there a way to modify the provided firmware version (except mfg_tool) ?

shubhamdp, 2024-01-17:
@a-tobias please take a look at this FAQ - https://docs.espressif.com/projects/esp-matter/en/latest/esp32/faq.html#a1-12-firmware-version-number

a-tobias, 2024-01-17:
As I was researching in a different direction the whole time, I simply didn't see it.

Thanks !

closed 2024-01-17 by a-tobias