espressif/esp-matter #801: LIGHT example: Failure to Commission via THEAD to Apple HomeKit & Google Home

nichonaugle, 2024-01-17:
**Describe the bug**
When testing the light example, the thread fails to connect to ANY ecosystem. No errors pop up and the example compiles without error, however fails to connect when commissioning.

**Environment**
- ESP-Matter Commit Id: master (current)
- ESP-IDF Commit Id: v5.1.2
- SoC (eg: ESP32 or ESP32-C3): ESP32-C6
- Device Logs (Please attach the log file): n/a
- Host Machine OS: MacOS (Commissioned via iPhone 15 Pro Max)
- Commissioner app and versions if present: HomeKit (newest using HomePod Mini) & Google Home (newest)
- Commissioner's logs if present: n/a

When attempting to commission the LIGHT example via Thread, it fails to do so and just shows the issue "Failed to pair. Please try again". It doesn't appear to be a BLE issue or a Wi-Fi issue, as the wi-fi example works without error on the same chip tested. I have also tried swapping between 3 different chips and continue to face the same issue. Compiled using different ESP-IDF commits as well and still face the same issue.

I erased everything and redownloaded fresh following the link below, however I was still unsuccessful:
Followed instructions here: https://docs.espressif.com/projects/esp-matter/en/latest/esp32c6/developing.html

I would also like to add that long term Thread network effects from the old driver software have been unreliable as they have been dropping connection and failing to reconnect. I understand that this may have been fixed in the new ESP-IDF commit, however I have been running a month long test using 7 ESP32-C6 examples loaded with different endpoint examples all running on Thread.

Please let me know if there is a different version I should be using or if the problem can be replicated on your end. Thanks in advance.

nichonaugle, 2024-01-17:
Also noticed that the two separate partition tables that used to be available in the light example (one for Thread and one for Wi-Fi) have been consolidated into one. I don't know if that is the root cause of the issue or if it is something else

shripad621git, 2024-01-18:
@nichonaugle ,I have tried the light example on the `latest esp-matter main branch` using `IDF version v5.1.2 `and the `commissioning over thread` worked fine in case of `apple home (version 17.2)`.Can you please share your `sdkconfig  file` and `device logs` in case of `commissioning over thread.`

nichonaugle, 2024-01-19:
Strange... I deleted and redownloaded the latest esp-matter and esp-idf v5.1.2 repositories and it seems to be working. 
previous controller lib commit: [79c83ca]
current controller lib commit: [77d09ce]
That is the only apparent change that was made so it appears that the issue is resolved.

closed 2024-01-19 by nichonaugle