espressif/esp-matter #802: Power Source Description buffer length is wrong.

jonsmirl, 2024-01-18:
[image]
[image]
[image]

In general, the declaration of the char[] buffers should use the constant and not a numeric size.

[image]
[image]

This issue occurs multiple times.

jonsmirl, 2024-01-18:
I changed this from 61 to 60 and now esp-matter doesn't abort, but there is still some problem.
[image]
[image]

jonsmirl, 2024-01-18:
[image]

I can't get this work when using sizeof(config->node_label) which matches the field limit. I get an "Error 1" from the Android decode library, but the buffer does look ok on the wire. (Maybe Android lib has a bug when the string is not zero terminated and max length?).

If I change this to strlen(config->node_label) I can get it to work.

jonsmirl, 2024-01-19:
When you retrieve a max length string saved to flash, do you trim off the zero terminator before giving it to CHIP? If it is not max length you want to keep the zero terminator.

jonsmirl, 2024-01-20:
in esp_matter_nvs.cpp
[image]

What happens if I am setting the string to NULL to remove it? Does look like it gets removed. Then later in the code I fault trying to print the attribute, but I think that is because the state is messed up.

So instead of 'err = ESP-OK' if needs to check to see if the key is in storage and remove it.

jonsmirl, 2024-01-20:
As a test I added this in my S3 code:
[image]
I verified that this does not remove the attribute from NVS. Still there, should be gone or have zero length
[image]
And internal RAM state is messed, a report now on this attribute will cause GFP in ROM code.

Reboot and attribute is back to what is in the NVS.

eme-bennytang, 2024-02-06:
hi @jonsmirl ,

We are seeing exactly the same problem in node_label (basic information), where the default string is a null 32 bytes char array
[image]

We have create an issue in connectedhomeip (https://github.com/project-chip/connectedhomeip/issues/31793)
They suggest we should pass-in strlen instead of sizeof when creating the attribute.

However in our previous trials, the strlen method will be ok in first read, but we can't write it by chip-tool

We do have a quick workaround (call an esp matter update for that attribute after power up, and everything works fine afterwards)
But this seems a patching fix and just want to know the issue better.

Have you got any further progress on this?

eme-bennytang, 2024-02-07:
@jonsmirl 
Just letting you know, found this fix in esp-matter today  :)
https://github.com/espressif/esp-matter/commit/3c6600df3fd281f3228bd8f677db60f60a7b223c
I have tested a bit and it has no problem reading and writing

shubhamdp, 2024-03-08:
@jonsmirl Bounds related problem is fixed in https://github.com/espressif/esp-matter/commit/d481fa9cf1d09001b1bbd9cd59dbb7836bbb13fe

shubhamdp, 2024-04-25:
@jonsmirl  `nvs_set_blob()` won't let us store the nullptr, so if thats the case, erasing that key, please check https://github.com/espressif/esp-matter/commit/8943861e30b6f96c17dfc3a8986f517977dc0451. And with this I think, string related problems are resolved.