espressif/esp-matter #804: Reporting Power or Current Consumption with Matter

bilalmalik76, 2024-01-19:
Is it possible to report the power or current consumption of a device using Matter? If so, are there any supported clusters available for that purpose?

jonsmirl, 2024-01-20:
It is still only in draft.
https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/draft/electrical-measurement-cluster.xml

Pips801, 2024-01-20:
I was wondering, does ESP-Matter only add support for released features, rather than provisional? I see that the codebase for the Cluster is in CHIP now (`ElectricalEnergyMeasurement`) 

- https://github.com/project-chip/connectedhomeip/tree/master/src/app/clusters/electrical-energy-measurement-server
- https://github.com/project-chip/connectedhomeip/tree/master/data_model/clusters/ElectricalEnergyMeasurement.xml

While it is a provisional item and low-priority, I could see support for this specific cluster being more requested than others - it seems like a lot of people are very interested in having matter support energy measurement. Being able to implement the cluster and interact with it in CHIP-tool would help people get ahead of the curve with this feature, even if it won't be released until the next Matter release 1.3.

jonsmirl, 2024-01-20:
I don't have CSA access, but I suspect it is not in Matter 1.3. Matter 1.3 is due out shortly and the code is still in a draft folder in the CHIP source code. BTW, that electrical measurement code has been in CHIP since the initial import from Zigbee, it is not new.

Currently, the easiest solution is to implement a custom cluster under esp-matter that matches the spec for the Electrical Measurements cluster. Since you are implementing this yourself you can leave out everything except the couple of fields you need -- likely voltage, current, and power.  Look at the Rainmaker code to see how to implement a custom cluster. This is pretty easy to do since those attributes are just simple read-only numbers. Implementing arrays and structs is much harder.

Of course if you do this be aware that there is no support in Apple, Google, or Amazon for reading Electrical Measurements so the only way you can see the data is with a vendor app.

bilalmalik76, 2024-01-21:
@jonsmirl We are not making our own app and want to use native Apple, Google, or Amazon ecosystem apps therefore custom clusters are not suitable for us. 
From this link, it looks like it is part of the 1.3 release: 
https://github.com/orgs/project-chip/projects/72/views/1?sliceBy%5Bvalue%5D=1.3+Feature+Complete&pane=issue&itemId=43595305

jonsmirl, 2024-01-21:
I have little faith this will be in 1.3. Even if it is, it might years until Google/Apple/Amazon implement it.  Google and Amazon still haven't implemented scene buttons and they were in the 1.0 spec.

Pips801, 2024-01-24:
> I have little faith this will be in 1.3. Even if it is, it might years until Google/Apple/Amazon implement it.  Google and Amazon still haven't implemented scene buttons and they were in the 1.0 spec.

This definitely seems to be the case - I'm very excited to see any support for 1.2, but it seems like there's very little communication or push from Apple or Google to get their app-side updates out, despite having advanced access to the 1.2 spec well before it was released, and being founding members of Matter/CSA.

With all of their involvement in making Matter happen, it seems a touch odd how little they want to continue the support of newer versions. Maybe it'll be in iOS 17.4, but I wouldn't hold my breath for it to happen any time soon. I believe in the CSAs ability to update and refine the spec and the chip framework way more than I believe in the implementers to carry those features into a usable app.

It's also got to be frustrating for manufacturers who are pouring all of this time and energy and money into developing Matter products like fridges, vacuums, Smoke alarms, etc just to get the cold shoulder from the companies who pushed for everyone to adopt Matter.

jonsmirl, 2024-01-24:
You also need to consider Google/Apple/Amazon's position here a little. Their objective is only to provide control of the basic elements of the device, not to provide a full UI for setting them up.  So for a light bulb they support turning it on/off but then they don't support setting the on/off ramp times. If you want to set those ramp times, that has to be done through another app, commonly from the vendor.

Based on this philosophy I have little faith GAA will ever add power monitoring support to their apps. However, they should add scene button and I don't know why Google hasn't (Apple has). The whole point of a scene button is to send a signal back to the controller where it will run a script.

So what support would GAA add for a vacuum? I don't see much which would be useful. Maybe a "Clean Now" voice command and then status for scripts to use -- cleaning active, bin full, charging, etc. They are never going to build a full UI, the vendor has to do that.

Now having said all of this... the full Matter API is available from the scripting engines of their controllers. So if you don't want to write a vendor app maybe you could write a bunch of scripts to achieve what you need.

About a year ago I tried to talk Apple into creating plugins for Apple Home. When Apple Home saw a new device it would load the vendor plugin which has full control of the device. That went nowhere. They don't want to dilute the quality perception of their app via plugins which might be buggy.

dhrishi, 2024-07-01:
@bilalmalik76 The electrical measurement clusters are a part of v1.3 and added to the esp-matter codebase.

bilalmalik76, 2024-07-01:
@dhrishi is Apple/Google home apps also have support for these clusters?

dhrishi, 2024-07-01:
@bilalmalik76 I don't think so, at this time. You can give it a try to verify.

closed 2024-07-01 by dhrishi