espressif/esp-matter #806: TC-SC-5.2 failing with group id 0x0101 (and only)

PhLuReh, 2024-01-22:
**Describe the bug**
When running TC-SC-5.2 on an official Matter TH, the test case fails at step `TH sends ViewGroup command` with `status value mismatch: expected 0 but got 139`. Using group id 0x0103 in the other steps works flawlessly.

DUT logs do not show any obvious errors. Adding and viewing groups with commands works, except for the tested id "0x0101".

Any hints?

**Environment**
- ESP-Matter Commit Id: dd4f34ec98ee1d5467bcfa8f6c885d53de58e50e
- ESP-IDF Commit Id: e088c3766ba440e72268b458a68f27b6e7d63986
- SoC (eg: ESP32 or ESP32-C3): ESP32
- Device Logs (Please attach the log file):
- Host Machine OS: `Linux version 6.1.29-1-MANJARO (builduser@fv-az292-908) (gcc (GCC) 12.2.1 20230201, GNU ld (GNU Binutils) 2.40) #1 SMP PREEMPT_DYNAMIC Wed May 17 14:00:55 UTC 2023`
- Commissioner app and versions if present: TH Version: v2.8.1-official | TH SHA: 1446f09
- Commissioner's logs if present: 
[TC-SC-5_2_2024_01_22_14_29_36.zip](https://github.com/espressif/esp-matter/files/14010733/TC-SC-5_2_2024_01_22_14_29_36.zip)

**Any additional details**
Test also failed at official testing offing for licencing.

PhLuReh, 2024-01-22:
I came across https://github.com/project-chip/connectedhomeip/issues/29352 added the Groups.Endpoint = 0 thing and retestet. Still failing. 
[TC-SC-5_2_2024_01_22_14_44_28.zip](https://github.com/espressif/esp-matter/files/14010869/TC-SC-5_2_2024_01_22_14_44_28.zip)

PhLuReh, 2024-01-23:
Of course, here is the TH report: [TC-SC-5_2_2024_01_22_14_44_28-74.zip](https://github.com/espressif/esp-matter/files/14020122/TC-SC-5_2_2024_01_22_14_44_28-74.zip)
DUT log is full of device-output (and even CAN-over-UART-bytes) and therefore hard to read. If you still want need it, I will retest and save the logs...

VaishaliAvhale, 2024-01-30:
@PhLuReh I went through the logs shared by you  in that I can see step's are mismatching but still the step ./chip-tool groupkeymanagement key-set-remove "0x01a3" 1 0 should work.

- I tried it at my end with same esp-matter commit id and it is working as expected! 

- Have you selected GroupKeyManagement PICS and Groups Cluster PICS while executinng this TC?

- While generating PICS make sure that you have select mandatory Feature, Attribute and commands required for Groups Cluster otherwise step related feature or attribute or command skip while execution.

PhLuReh, 2024-02-09:
I try to answer your questions.

GRPKEY.S (Does the device implement the Group Key Management cluster as a server?) is marked as supported (true).
G.S (Does the device implement the Groups cluster as a server?) is marked as supported (true).

I rechecked the PICS, which I use in the Matter TH with the online [PICS tool](https://picstool.csa-iot.org/), and there's no obvious error related to this test case:
```
General Diagnostics Cluster Test Plan.xml: Validated 26 items with 0 warnings and 0 errors.
Group Communication.xml: Validated 6 items with 0 warnings and 0 errors.
Groups Cluster Test Plan.xml: Validated 22 items with 0 warnings and 0 errors.
Identify Cluster Test Plan.xml: Validated 17 items with 1 warning and 0 errors.
Multiple Fabrics Test Plan.xml: Validated 18 items with 0 warnings and 0 errors.
OTA Software Update Test Plan.xml: Validated 4 items with 0 warnings and 0 errors.
Scenes Cluster Test Plan.xml: Validated 35 items with 0 warnings and 0 errors.
Wi-Fi Network Diagnostics Cluster Test Plan.xml: Validated 36 items with 0 warnings and 0 errors.
Window Covering Cluster Test Plan.xml: Validated 82 items with 1 warning and 2 errors.
Access Control cluster.xml: Validated 11 items with 0 warnings and 0 errors.
Access Control Enforcement Test Plan.xml: Validated 1 item with 0 warnings and 0 errors.
Base.xml: Validated 179 items with 2 warnings and 0 errors.
Descriptor Cluster TestPlan.xml: Validated 10 items with 0 warnings and 0 errors.
General Commissioning Cluster Test Plan.xml: Validated 13 items with 0 warnings and 0 errors.
Device Management Test Plan.xml: Validated 126 items with 2 warnings and 0 errors.
```
[the 2 relevant PICS files](https://github.com/espressif/esp-matter/files/14222714/issue806_pics.zip)

I would like to try to run it manually. could you provide the chip-tool calls for this? I tried, but I failed already at the second and third step. How to write the ACLs?
`./chip-tool accesscontrol write acl '[{"fabricIndex": 1,"privilege": 5,"authMode": 2,"subjects": 1,"targets": null },{"fabricIndex": 1,"privilege": 4,"authMode": 3,"subjects": [259],"targets": null}]'  1 0`

Thanks in advance.

VaishaliAvhale, 2024-02-21:
I have checked your PICS file and it is correct. Please find the attached TC-SC-5.2-steps file for manual test execution. I have also attached the device and chip-tool logs of TC-SC-5.2, which I have executed on my end with the same specifications used by you. I am getting the expected output for all the steps. Please refer to the following links for the logs and test case document:

[TC-SC-5-2-chip-tool.log](https://github.com/espressif/esp-matter/files/14358247/TC-SC-5-2-chip-tool.log)
[TC-SC-5-2-DUT.log](https://github.com/espressif/esp-matter/files/14358249/TC-SC-5-2-DUT.log)
[TC-SC-5.2-Steps.ods](https://github.com/espressif/esp-matter/files/14358250/TC-SC-5.2-Steps.ods)

PhLuReh, 2024-02-21:
Don't get me wrong, I'm very happy about your support, but there's some quirks going on...

I manually run the steps from your [TC-SC-5.2-Steps.ods](https://github.com/espressif/esp-matter/files/14358250/TC-SC-5.2-Steps.ods):
```
./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null },{"fabricIndex": 1, "privilege": 4, "authMode": 3, "subjects": ["0x0103"], "targets": null }]' 1 0 > 2024-02-21-TC-SC-5_2.log
./chip-tool groupkeymanagement key-set-write '{"groupKeySetID": "0x01a3","groupKeySecurityPolicy": 0, "epochKey0":"d0d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime0": 1,"epochKey1":"d1d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime1": 18446744073709551613,"epochKey2":"d2d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime2": 18446744073709551614 }' 1 0 >> 2024-02-21-TC-SC-5_2.log
./chip-tool groupkeymanagement write group-key-map '[{"groupId": "0x0103", "groupKeySetID": "0x01a3", "fabricIndex": 1},{"groupId": "0x0101", "groupKeySetID": "0x01a3", "fabricIndex": 1}]' 1 0 >> 2024-02-21-TC-SC-5_2.log
./chip-tool groups remove-all-groups 1 0 >> 2024-02-21-TC-SC-5_2.log
```

The fourth call was unsuccessful. I then had a look at your [TC-SC-5-2-chip-tool.log](https://github.com/espressif/esp-matter/files/14358247/TC-SC-5-2-chip-tool.log), where you ran:
```
./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null },{"fabricIndex": 1, "privilege": 4, "authMode": 3, "subjects": ["0x0103"], "targets": null }]' 1 0
./chip-tool groupkeymanagement key-set-write '{"groupKeySetID": "0x01a3","groupKeySecurityPolicy": 0, "epochKey0":"d0d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime0": 1,"epochKey1":"d1d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime1": 18446744073709551613,"epochKey2":"d2d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime2": 18446744073709551614 }' 1 0
./chip-tool groupkeymanagement write group-key-map '[{"groupId": "0x0103", "groupKeySetID": "0x01a3", "fabricIndex": 1},{"groupId": "0x0101", "groupKeySetID": "0x01a3", "fabricIndex": 1}]' 1 0
./chip-tool groups remove-all-groups 1 1
```

I also ran the `remove-all-groups` on endpoint `1` like you did, which also succeeded. Can you run your provided steps, or does it also fail on endpoint `0`? 

I continued running the commands from your `chip-tool.log` and all of them succeed. So is the test steps from the chip-tool wrong or what is the clue?

[2024-02-21-TC-SC-5_2-DUT.log](https://github.com/espressif/esp-matter/files/14360372/2024-02-21-TC-SC-5_2-DUT.log)
[2024-02-21-TC-SC-5_2-chip-tool.log](https://github.com/espressif/esp-matter/files/14360371/2024-02-21-TC-SC-5_2-chip-tool.log)

These were the remaining commands I ran with success:
```
./chip-tool groups remove-all-groups 1 1 >> 2024-02-21-TC-SC-5_2.log
./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null }]' 1 0 >> 2024-02-21-TC-SC-5_2.log
./chip-tool groups add-group "0x0103" "Test Group 0103" 1 0 >> 2024-02-21-TC-SC-5_2.log #failing as well
./chip-tool groups add-group "0x0103" "Test Group 0103" 1 0 >> 2024-02-21-TC-SC-5_2.log #failing as well
./chip-tool groups add-group "0x0103" "Test Group 0103" 1 1 >> 2024-02-21-TC-SC-5_2.log
./chip-tool groups add-group 257 "Test Group 0101" 0xffffffffffff0103 1 --trace_decode 1 >> 2024-02-21-TC-SC-5_2.log
./chip-tool groups view-group "0x0101" 1 1 >> 2024-02-21-TC-SC-5_2.log
./chip-tool groupkeymanagement key-set-remove "0x01a3" 1 0 >> 2024-02-21-TC-SC-5_2.log
./chip-tool groups remove-all-groups 1 1 >> 2024-02-21-TC-SC-5_2.log
./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null }]' 1 0 >> 2024-02-21-TC-SC-5_2.log
```

VaishaliAvhale, 2024-02-21:
I continued running the commands from your chip-tool.log and all of them succeed. So is the test steps from the chip-tool wrong or what is the clue?
>>> 
Groups Cluster is a application cluster and it is present on endpoint 1 so that the command is working on endpoint 1 and getting Unsupported Response on endpoint 0 which is expected behaviour.
- All steps are executing successfully means issue not on application side. You need to check on TH side. Do you have TestHarness USER GUIDE?
Note: 
1. I have send you the verification document steps and endpoint may vary as per implementation.

PhLuReh, 2024-02-22:
@VaishaliAvhale I don't get the point. What is my task? Can you edit your comment and mark citations/replies, so that it is more clear?

I agree in groups cluster is application-specific and therefore should be run on endpoint 1+. I'm not the official test-lab, so you would argue, the test-lab needs to know that it should use endpoint 1 in favour of 0?

VaishaliAvhale, 2024-02-22:
You are executing this TC on TH right If TC-SC-5.2 Manually Executed with Success so there is no issue with the firmware please try it with TH and map the command with Manual test logs and if endpoint is wrong for groups cluster raise issue on test harness repo.

PhLuReh, 2024-02-22:
Discussing this with my official test lab and will give feedback here. Thx for now.

dhrishi, 2024-03-12:
@PhLuReh Any update to share? Please close this issue if resolved

PhLuReh, 2024-03-12:
No explicit feedback on this, but it looks like the test passed now... Closing for now.

closed 2024-03-12 by PhLuReh