espressif/esp-matter #808: Failed to read any factory data through ESP32FactoryDataProvider

eme-bennytang, 2024-01-23:
**Describe the bug**
I have configured in SDK config as below
[image]
While its ok to read out the DAC (via ESP32SecureCertDACProvider) for checking, I failed in reading the factory data (via ESP32FactoryDataProvider)

**Environment**
- ESP-Matter Commit Id: 2f692ac
- ESP-IDF Commit Id: tag v5.1.2
- SoC (eg: ESP32 or ESP32-C3): ESP32S3
- Host Machine OS:Linux

**Any additional details**
I added some additional log
[image]
And the result is always ESP_ERR_NVS_NOT_FOUND

I did the same thing in esp-idf v5.1 with matter 548ef58 and it works perfectly fine
But after my stack updates to the current version, it failed and stuck in the mentioned step

eme-bennytang, 2024-01-24:
I have found the issue. 
Somehow the nvs encryption is default ON.
Once I have set CONFIG_NVS_ENCRYPTION=n in my SDK config, then it is resolved.

But here's another question:
[image]
As per my understanding, even with nvs encryption enabled, the MCU will only read/write as encrypted info when that partition is set as "encrypted"
I didnt mark my nvs_factory partition as encrypted, why does it assume I have?

shripad621git, 2024-01-31:
@eme-bennytang ,you can find the appropriate information for using `nvs_encryption` [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_encryption.html) .

shripad621git, 2024-01-31:
@eme-bennytang , please close the issue if it is resolved.

eme-bennytang, 2024-02-06:
hi @shripad621git 
I do have more understanding for nvs_encryption now after some testing and have solved my problem.

However I have another question:
With the factory data generation tools (mfg_tool.py), I can generate an encrypted factory data along with corresponding nvs_encryption key.
Is it possible I can use this generated key to encrypt some other data that needs to be read from other nvs partition?

shripad621git, 2024-02-27:
@eme-bennytang , I would Iike to know the extra data you wish to encrypt in the nvs_partition.  If you wish to encrypt some extra data in nvs , you can use the `csv and mcsv` option in `mfg_tool.`You can refer [here](https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool#generate-factory-partitions-with-extra-nvs-key-values-specified-using-csv-and-mcsv-file-optional-arguments----csv-and---mcsv). The `--encrypt ` option in mfg_tool will encrypt the data in nvs partition.

closed 2024-02-27 by eme-bennytang