espressif/esp-matter #809: ESP32 door lock example crashing

bilalmalik76, 2024-01-23:
**Describe the bug**
ESP32 door lock example available in connected home IP example causes ESP32 to crash and reboot when I press the door lock button:
[image]

bilalmalik76, 2024-01-23:
Even if I don't use ScheduleWork to execute UpdateClusterState and call it directly still get this error: 
chip::DeviceLayer::SystemLayer().ScheduleWork(UpdateClusterState, nullptr);

error by calling UpdateClusterState function directly:
[image]

bilalmalik76, 2024-01-23:
@shubhamdp 
Environment

ESP-Matter Commit Id: 01bebca56b42a35b5ad049f59292df7d4c08ea6b
ESP-IDF Commit Id: cbce221e88d52665523093b2b6dd0ebe3f1243f1
SoC (eg: ESP32 or ESP32-C3):  ESP32
Device Logs (Please attach the log file):
Host Machine OS: Linux
Commissioner app and versions if present: Apple home app
Commissioner's logs if present: NA

dhrishi, 2024-01-24:
@bilalmalik76 If you are trying out the connectedhomeip lock-app example, then can you confirm if you have this commit: https://github.com/project-chip/connectedhomeip/pull/31500 . If not, can you cherry-pick the commit and try?

The connectedhomeip submodule in esp-matter repo is not updated to use this commit yet

bilalmalik76, 2024-01-24:
@dhrishi I think you didn't get the reason for the application creash and assumed that it is related to another issue. This issue is not related to Factory Reset Triggered. The commit you shared fixes Factory Reset Triggered. https://github.com/project-chip/connectedhomeip/pull/31500

My issue is different whenever this UpdateClusterState function executes it crashes application and reason for crashing is:
[image]

and if you see this ChipLogError is being used in the following function. I think, in your sample application, UpdateClusterState function is not being called correctly. 
[image]

dhrishi, 2024-01-25:
@bilalmalik76 Yes, my bad. Similar fix was done some time back https://github.com/project-chip/connectedhomeip/pull/31171. Can you please try with project-chip/connectedhomeip latest master branch directly instead of esp-matter connectedhomeip submodule (not updated to the latest yet) so that we are sure that you have all the required fixes.

dhrishi, 2024-01-30:
@bilalmalik76 Is there any update you can share

bilalmalik76, 2024-01-30:
Same issue with project-chip/connectedhomeip latest master. @dhrishi can you please try on your side?

bilalmalik76, 2024-01-31:
@dhrishi This error occurs only when I try to lock and unlock from esp32 button, not from app. Can you share how did you send unlock-door and lock-door commands?

dhrishi, 2024-01-31:
@bilalmalik76 The screenshot you've added in [this comment](https://github.com/espressif/esp-matter/issues/809#issuecomment-1906593608) shows that it failed while you were controlling it from an ecosystem Matter app or chip-tool. See the prints of Invoke command response. These should not be seen if you are controlling it through the button. Please let me know if I am missing anything.

I also tried changing the state with button. Note that I had to change the lock/unlock GPIO to 0 (`APP_LOCK_BUTTON`) as per my ESP32 devkit in `examples/platform/esp32/lock/AppConfig.h`. Please see the logs for lock, unlock below -

```
I (15665) lock-app: Lock Action has been initiated
I (17665) lock-app: Lock Action has been completed
I (17665) chip[ZCL]: 0x3f40e948 ep 1 clus 0x0000_0006 attr 0x0000_0000 not supported 
I (17665) lock-app: ERR: updating on/off c3 # Ignore this as cluster 6 (OnOff) on ep 1 is not there
I (26885) lock-app: Unlock Action has been initiated
I (28885) lock-app: Unlock Action has been completed
```

dhrishi, 2024-02-06:
@bilalmalik76 Any update to share? Please close the issue if it works fine

bilalmalik76, 2024-02-10:
@dhrishi Yes I changed the lock/unlock GPIO according to my dev kit but it is still crashing even after using latest commit of connect home ip.

dhrishi, 2024-02-11:
This should not happen. Please share the connectedhomeip latest commit id you are using

dhrishi, 2024-02-26:
Please provide the exact commit id used and re-open the issue.

closed 2024-02-26 by dhrishi