espressif/esp-matter #810: How to delete features in a cluster

MaplestoryAlen, 2024-01-25:
I am currently facing a problem.
   For example, if I add the auto mode of the thermostat, how can I delete it,when turning on auto mode, it will affect the minimum and maximum temperature setting of the cooling/heating mode, so I want to remove the auto mode function when switching to cooling/heating mode

code:
    cluster::thermostat::feature::auto_mode::config_t auto_mode_config;
    cluster::thermostat::feature::auto_mode::add(cluster, &auto_mode_config);

wqx6, 2024-01-26:
AutoMode feature is an optional feature for thermostat cluster, so we don't add it to thermostat cluster by default. You can add it if you device supports that feature.

But the features for a device should be determined when creating the device and never changed. It is odd for a device which supports a feature at some times and doesn't support the same feature at other times.

MaplestoryAlen, 2024-01-29:
> AutoMode feature is an optional feature for thermostat cluster, so we don't add it to thermostat cluster by default. You can add it if you device supports that feature.
>
> But the features for a [...]

For example, the minimum set temperature for refrigeration and heating is 16 degrees Celsius, and the maximum temperature is 32 degrees Celsius. After turning on the auto mode and switching to a separate refrigeration and heating mode, when using the chip tool to set the maximum refrigeration and heating temperature to 16 degrees Celsius, an error will be reported and error code 0x87 will be returned

The system has three modes of auto cooling and heating

MinHeatSetpointLimit = 16 
MaxHeatSetpointLimit = 32
MinCoolSetpointLimit = 16 
MaxCoolSetpointLimit = 32
MinSetpointDeadBand = 2.5

OccurpiedHeatingSetpoint= 24，set the cooling range to 26.5-29.5 degrees Celsius
OccupideCoolingSetPoint = 27,   set the heating range to 22~24.5 degrees Celsius

So what I need is a separate cooling or heating mode, which can be set to a minimum temperature of 16 and a maximum temperature of 32

jadhavrohit924, 2024-01-29:
@MaplestoryAlen This is expected behaviour, please check [this](https://github.com/espressif/esp-matter/issues/773#issuecomment-1867535043). If you don’t need auto you can remove it at the time of creation of device

closed 2024-01-30 by MaplestoryAlen