espressif/esp-matter #813: WSL distribution cannot be reached via the WSL virtual switch

oidebrett, 2024-01-27:
I am trying to set up my Windows 11 PC for ESP32 matter development using WSL2. I am following the instructions as per the [Programming Guide - Working with the CHIP Tool in WSL2](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/using_chip_tool.html).

After creating the Hyper-V virtual switch and changing my .wslconfig file, I can successfully receive mDns advertisements from the local network.

However, this set up seems to have prevented me from attaching USB devices (I.e. ESP32 DK) via WSL. The command I use is below. When I remove the virtual switch from the Hyper-V manager and also return my .wslconfig to its original state - I can attach the USB devices.

### Out from usbipd command in powershell
`
PS C:\WINDOWS\system32> usbipd wsl list
BUSID  VID:PID    DEVICE                                                        STATE
1-1    10c4:ea60  CP2102 USB to UART Bridge Controller                          Not attached
1-5    1286:204c  Marvell AVASTAR Bluetooth Radio Adapter                       Not attached
1-7    045e:07de  USB Input Device                                              Not attached
1-16   045e:0306  Realtek USB 3.0 Card Reader                                   Not attached

PS C:\WINDOWS\system32> usbipd wsl attach -d Ubuntu-22.04 -b 1-1
usbipd: error: The selected WSL distribution cannot be reached via the WSL virtual switch; try restarting the WSL distribution.
` 

### My .wslconfig file

`
[wsl2]
kernel=c:\\Users\\admin\\bluetooth-bzImage 

memory=6GB 

localhostForwarding=true 

networkingMode = bridged 

vmSwitch = Bridge 

ipv6 = true
`

If this is a known issue and there is a work around, could the documentation be updated? I am happy to make a PR for this.

oidebrett, 2024-01-27:
I found a fix for this as outlined in the [usbipd-win repo](https://github.com/dorssel/usbipd-win/issues/136#issuecomment-1754007527)

> The logic was made by Microsoft. It looks for a (single) switch with WSL in the name.

I got the "usbipd wsl attach" command to work by changing the name of the vmSwitch to "WSL Bridge" instead of "Bridge". Change the name in both Hyper-v manager and in the .wslconfig file.

It seems that the "usbipd wsl attach" command looks for virtual switches with the word "WSL" in them. Strange but apparently true!

A simple change to the doco on this [page](https://github.com/espressif/esp-matter/blob/6ad5710c/docs/en/using_chip_tool.rst) would fix this.

sayondeep, 2024-01-29:
Hello @oidebrett ,
I am not able to reproduce the issue and have a working setup with bridge name setup as "Bridge".
To resolve the issue can you please help me with the following details:
output of `wsl --version` and `uspipd --version`

oidebrett, 2024-01-29:
Hi,

Thanks for looking into this. Heres the information requested.

```
PS C:\WINDOWS\system32> wsl --version
WSL version: 2.0.9.0
Kernel version: 5.15.133.1-1
WSLg version: 1.0.59
MSRDC version: 1.2.4677
Direct3D version: 1.611.1-81528511
DXCore version: 10.0.25131.1002-220531-1700.rs-onecore-base2-hyp
Windows version: 10.0.22631.3007
```

```
PS C:\WINDOWS\system32> wsl --list --all --verbose
  NAME            STATE           VERSION
* Ubuntu-22.04    Running         2
PS C:\WINDOWS\system32> usbipd --version
2.4.1+1.Branch.master.Sha.90acf9456020ca8c6310ca62a71ee23cb6ca34ad

```

```

PS C:\WINDOWS\system32> ipconfig

Windows IP Configuration

Ethernet adapter vEthernet (WSL Bridge):

   Connection-specific DNS Suffix  . : home
   IPv6 Address. . . . . . . . . . . : 2001:bb6:46b5:6c00:d93:8e88:4246:ec23
   Temporary IPv6 Address. . . . . . : 2001:bb6:46b5:6c00:61e8:2d15:e2b8:3966
   Link-local IPv6 Address . . . . . : fe80::3a99:8f20:eb15:c0ad%21
   IPv4 Address. . . . . . . . . . . : 192.168.1.9
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Default Gateway . . . . . . . . . : fe80::66fd:96ff:fef1:dfc2%21
                                       192.168.1.254

Wireless LAN adapter Local Area Connection* 1:

   Media State . . . . . . . . . . . : Media disconnected
   Connection-specific DNS Suffix  . :

Wireless LAN adapter Local Area Connection* 2:

   Media State . . . . . . . . . . . : Media disconnected
   Connection-specific DNS Suffix  . :

Ethernet adapter Bluetooth Network Connection:

   Media State . . . . . . . . . . . : Media disconnected
   Connection-specific DNS Suffix  . :

Ethernet adapter vEthernet (Default Switch):

   Connection-specific DNS Suffix  . :
   Link-local IPv6 Address . . . . . : fe80::19be:c51a:f746:cab9%17
   IPv4 Address. . . . . . . . . . . : 172.27.112.1
   Subnet Mask . . . . . . . . . . . : 255.255.240.0
   Default Gateway . . . . . . . . . :
```

On my WSL instance

```
$ uname -r
6.1.21.2-microsoft-standard-WSL2+
```

Yes - I have the path set correctly for the kernel. 

```
[wsl2]
kernel=C:\\Users\\admin\\bluetooth-bzImage
memory=6GB
localhostForwarding=true
networkingMode = bridged
vmSwitch = WSL Bridge
ipv6 = true
```

sayondeep, 2024-01-29:
Thanks @oidebrett, I am looking into the issue.
I see that you are using "Marvell AVASTAR Bluetooth Radio Adapter " for bluetooth.
It would really help to know if you are able to scan the bluetooth devices from WSL using `bluetoothctl scan on`. 
or even better are you able to commission matter devices using chip-tool from WSL instance.

sayondeep, 2024-01-29:
@oidebrett 
Thanks for the information provided. I was able to reproduce the issue. The issue is with the version of usbipd being used.
When WSL2 instance is started with default config a vmswitch named WSL is created and it persists until the system is reboot(can be seen on Hyper-V). During this even if we create and configure new vmswitch usbipd still finds "WSL" vmswitch and tries to use that. This "WSL" named switch is removed after reboot.

> I got the "usbipd wsl attach" command to work by changing the name of the vmSwitch to "WSL Bridge" instead of "Bridge".

You may be correct about using this naming for this version of usbipd. On naming it "WSL Bridge" it worked (only when "WSL" vmswitch was not present). But on naming it "Bridge" it gave error : `usbipd: error: The local IP address for the WSL virtual switch could not be found.` even when "WSL" named vmswitch was not there.

Conclusion, more than one vmswitch containing "WSL" in its name cause problem with this version of usbipd.

Can you please update usbipd using `winget install usbipd`. and use the following commands to bind and attach:
Please try naming the vmswitch as Bridge(for the purpose of testing).
```
usbipd list
usbipd bind --busid=<BUSID>
usbipd attach --wsl --busid=<BUSID>
```
The commands have changed with the new update.We will update the docs accordingly. Thanks for reporting the issue.

oidebrett, 2024-01-29:
Hi,
Thanks for replying.

I tried your suggested fixed but unfortunately I cant get it working at all.

I upgraded usbipd to the latest using:
```
winget install usbipd
```

now the version is
```
PS C:\WINDOWS\system32> usbipd --version
4.0.0+182.Branch.master.Sha.2ffe37ec799b9e73eb9d23d051d980fefb616ce1
```

But now I cant  attach the device at all and I get an error regarding a TCP connection. Pinging the wsl instance works but the usbipd attach --wsl --busid=<BUSID> command fails.

``` 
PS C:\WINDOWS\system32> usbipd --version
4.0.0+182.Branch.master.Sha.2ffe37ec799b9e73eb9d23d051d980fefb616ce1
PS C:\WINDOWS\system32> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
1-1    1d57:fa60  USB Input Device                                              Not shared
1-5    1286:204c  Marvell AVASTAR Bluetooth Radio Adapter                       Shared
1-7    045e:07de  USB Input Device                                              Not shared
1-16   045e:0306  Realtek USB 3.0 Card Reader                                   Not shared

Persisted:
GUID                                  DEVICE
24a082e1-9e53-43b6-ab36-74ccadc0735e  Dual RS232-HS
565e13c0-0bc5-4e35-9ddd-a25c75803e8f  USB Serial Device (COM3), USB JTAG/serial debug unit
754049ff-17a1-4511-9d40-ce4f4cced579  CP2102N USB to UART Bridge Controller
c7a28df7-4fed-4d60-9d9f-8386f61479a2  CP2102 USB to UART Bridge Controller
de754873-3ecf-44f9-91fa-07b9128d6a3b  CP2102N USB to UART Bridge Controller

PS C:\WINDOWS\system32> usbipd bind --busid=1-5
usbipd: info: Device with busid '1-5' was already shared.
PS C:\WINDOWS\system32> usbipd attach --wsl --busid=1-5
usbipd: info: Using WSL distribution 'Ubuntu-22.04' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.16.7.81 to reach the host.
WSL usbip: error: tcp connect
usbipd: error: Failed to attach device with busid '1-5'.
PS C:\WINDOWS\system32> ping 172.16.7.81

Pinging 172.16.7.81 with 32 bytes of data:
Reply from 172.16.7.81: bytes=32 time<1ms TTL=128

Ping statistics for 172.16.7.81:
    Packets: Sent = 1, Received = 1, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 0ms, Maximum = 0ms, Average = 0ms
```

another strange result of upgrading usbipd and going back to "Bridge" as the name of the virtual switch is that domain name resolution no longer works from my wsl instance

from my wsl instance I get
```
ivob@DESKTOP-J0S44C6:~$ ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
^C
--- 8.8.8.8 ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms

ivob@DESKTOP-J0S44C6:~$ ping github.com
ping: github.com: Temporary failure in name resolution
```

I am using a Surface Pro 6 - if that makes any difference. I am going to try and downgraded usbipd to version 2.4.1 and see if I can get it working again using the fix that I suggested above. Please let me know if you have other suggestions on how I can get the latest usbipd working.

sayondeep, 2024-01-29:
@oidebrett The problem here is that you have been assigned the same ip address as host by the hyper-V. It is a known problem. Please see [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/using_chip_tool.html#troubleshoot).

oidebrett, 2024-01-29:
Hi Sayondeep,

I got it working in the way you suggested using the latest version of usbipd version 4.0.0+

The actual problem was related to the network that I was on which must be blocking tcp connections between hosts. I am now on my original wifi network and it is working as you suggested. So an update of the documentation for the usbipd commands is all that is required. 

Many thanks

sayondeep, 2024-01-29:
> Thanks @oidebrett, I am looking into the issue. I see that you are using "Marvell AVASTAR Bluetooth Radio Adapter " for bluetooth. It would really help to know if you are able to scan the bluetooth [...]

Can you please provide information regarding this, if possible.

oidebrett, 2024-01-29:
> Can you please provide information regarding this, if possible.

Yes. I can commission using the Marvell AVASTAR Bluetooth Radio Adapter " for bluetooth.

sayondeep, 2024-01-29:
Thankyou @oidebrett . Feel free to close the issue.

oidebrett, 2024-01-29:
Closed with resolution being update to new usbipd version