espressif/esp-matter #814: Matter example not building properly (VSCode WSL on Windows 11)

FloBer431, 2024-01-29:
**Describe the bug**
I installed and set up the whole VS Code - WSL2 environment like described in the guide.
I'm working on a Windows 11 system with Ubuntu on the WSL2.
On VS Code I'm using the WSL connect feature.

When opening the "Light" Example of the MatterSDK an dtrying to build it, I always get the same Error:

```
 *  Executing task: ninja  

[3/564] Performing build step for 'bootloader'
[1/1] cd /home/ber/.espressif/esp-matter/examples/light/build/bootloader/esp-idf/esptool_py && /home/ber/.espressif/python_env/idf5.3_py3.10_env/bin/python /home/ber/esp/esp-idf/components/partition_table/check_sizes.py --offset 0xc000 bootloader 0x0 /home/ber/.espressif/esp-matter/examples/light/build/bootloader/bootloader.bin
Bootloader binary size 0x5240 bytes. 0x6dc0 bytes (57%) free.
[6/564] Performing configure step for 'chip_gn'
FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure 
cd /home/ber/.espressif/esp-matter/examples/light/build/esp-idf/chip && /usr/bin/gn --root=/home/ber/.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32 gen --check --fail-on-unused-args /home/ber/.espressif/esp-matter/examples/light/build/esp-idf/chip && /usr/local/bin/cmake -E touch /home/ber/.espressif/esp-matter/examples/light/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-configure
ERROR at //build_overrides/pigweed_environment.gni:20:1: Unable to load "/home/ber/.espressif/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/build_overrides/pigweed_environment.gni".
import("${_bootstrap_root}/build_overrides/pigweed_environment.gni")
^------------------------------------------------------------------
See //third_party/connectedhomeip/third_party/pigweed/repo/pw_toolchain/clang_tools.gni:15:1: whence it was imported.
import("//build_overrides/pigweed_environment.gni")
^-------------------------------------------------
See //third_party/connectedhomeip/third_party/pigweed/repo/pw_toolchain/host_clang/toolchains.gni:18:1: whence it was imported.
import("$dir_pw_toolchain/clang_tools.gni")
^-----------------------------------------
See //third_party/connectedhomeip/third_party/pigweed/repo/pw_unit_test/test.gni:22:1: whence it was imported.
import("$dir_pw_toolchain/host_clang/toolchains.gni")
^---------------------------------------------------
See //third_party/connectedhomeip/third_party/pigweed/repo/pw_build/BUILD.gn:23:1: whence it was imported.
import("$dir_pw_unit_test/test.gni")
^----------------------------------
See //third_party/connectedhomeip/build/config/compiler/BUILD.gn:180:19: which caused the file to be included.
      configs = [ "$dir_pw_build:optimize_size" ]
                  ^----------------------------
[8/564] Linking C static library esp-idf/esp_wifi/libesp_wifi.a
ninja: build stopped: subcommand failed.

 *  The terminal process "/bin/bash '-c', 'ninja '" failed to launch (exit code: 1). 
```
If necessary I can also share the whole ouput given in the terminal.

I already read other issues with an similar problem, but nothing worked out for me.

**Any additional details**
Extra Information: My plan is to build my own device wich is an contact_sensor watching an gpio_input on the board and can then trigger automations in googlehome/alexa/apple home. Maybe somebody has done something like this already or know where to get an example for this.

FloBer431, 2024-01-30:
@sayondeep Sorry, but where can I find these commit ids, I dont't know where and how to find.
The installation worked fine for me.

Thank you very much, deleting and reinstalling it into `/home/ber` worked out fine for me!

But now I got into a new problem, using the `>ESP-IDF: Select port to use` command, I run into following Error: `Something went wrong while getting the serial port list` So I cannot flash the program onto my esp32 right now.

sayondeep, 2024-01-30:
@FloBer431 Can you  please confirm if you have attached the device to WSL instance using `usbipd` command ? Please see the 4th point [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#windows-10)
Is the device showing on running `ls -l /dev/ttyUSB*` inside WSL instance.

Please install the latest version of usbipd using `winget install usbipd`
Please use the following commands to attach the device to WSL instance
```
usbipd list
usbipd bind --busid=<BUSID>
usbipd attach --wsl --busid=<BUSID>
```

dhrishi, 2024-01-30:
@FloBer431 Please take a look at the [all_device_types example](https://github.com/espressif/esp-matter/tree/main/examples/all_device_types_app). `create --device_type` on the console will list the supported device types, which should have the contact sensor as well. You can give it a try. Code is [here](https://github.com/espressif/esp-matter/blob/main/examples/all_device_types_app/main/esp_matter_console_helpers.cpp#L257)

I am closing this issue as the original problem is solved.

closed 2024-01-30 by dhrishi