espressif/esp-matter #815: Expected behaviour in TC-SU-2.8

PhLuReh, 2024-01-30:
**Describe the question/query that you have**
My testing laboratory asks about the behaviour of my DUT in this test-case. What is the timeout for a unresponsive OTA provider in this case? Can you specify the esp-matter implementation?

**Additional context**
The question arose when executing [TC-SC-2.8](https://github.com/espressif/connectedhomeip/blob/v1.1.0.1/src/app/tests/suites/certification/Test_TC_SU_2_8.yaml) from the official testcases for v1.1 testing.

PhLuReh, 2024-02-29:
My device is still not succeeding in this test-case. I got the full command list from out test-lab to run it on the device. Everything works fine until the last step, when the (unavailable) OTA provider gets announced to the DUT: `./chip-tool otasoftwareupdaterequestor announce-otaprovider 0xDEADBEEF 0 0 0 0x1234567890 0`

There are some timeouts in the log:
```
I (185020) chip[DIS]: Checking node lookup status after 45005 ms
E (185020) chip[DIS]: OperationalSessionSetup[1:00000000DEADBEEF]: operational discovery failed: 32
E (185020) chip[SWU]: Failed to connect to node 0x00000000DEADBEEF: 32
I (185030) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000002A's Attribute 0x00000002 is 2 **********040) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x0000002A's Attribute 0x00000002 is 1 **********050) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000002A's Attribute 0x00000003 is null **********) chip[SWU]: Unknown idle state reason so set the periodic timer for a next attempt
I (185070) chip[SWU]: Stopping the watchdog timer
I (185080) chip[SWU]: Starting the periodic query timer, timeout: 86400 seconds
I (187370) chip[EM]: >>> [E:12647r S:0 M:195917867] (U) Msg RX from 0:B8E029A9EC613E6E [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)CASE Server received Sigma1 message . Starting handshake. EC 0x3ffc7d20
I (187390) chip[EM]: <<< [E:12647r S:0 M:125362571 (Ack:195917867)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)125362571 to IP address 'UDP:[FD61:3599:E3A1::1C2%st1]:59748'
I (187410) chip[SC]: Received Sigma1 msg
I (187420) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0x0000001234567890
I (187850) chip[EM]: <<< [E:12647r S:0 M:125362572 (Ack:195917867)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:31 (SecureChannel:CASE_Sigma2)g 125362572 to IP address 'UDP:[FD61:3599:E3A1::1C2%st1]:59748'
I (187870) chip[SC]: Sent Sigma2 msg
I (187880) chip[EM]: >>> [E:12647r S:0 M:195917867] (U) Msg RX from 0:B8E029A9EC613E6E [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)<<< [E:12647r S:0 M:125362573 (Ack:195917867)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)125362573 to IP address 'UDP:[FD61:3599:E3A1::1C2%st1]:59748'
I (187910) chip[EM]: >>> [E:12647r S:0 M:195917868 (Ack:125362572)] (U) Msg RX from 0:B8E029A9EC613E6E [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3)0 M:125362574 (Ack:195917868)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)125362574 to IP address 'UDP:[FD61:3599:E3A1::1C2%st1]:59748'
I (187950) chip[SC]: Received Sigma3 msg
E (187960) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time88640) app_main: High Water Marks
I (188640) app_main: - APP Task: 17ec
I (188640) app_main: - CAN Task: 169c
E (188950) chip[DL]: Long dispatch time: 990 ms, for event type 2
I (189010) chip[EM]: <<< [E:12647r S:0 M:125362575 (Ack:195917868)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport) 125362575 to IP address 'UDP:[FD61:3599:E3A1::1C2%st1]:59748'
I (189020) chip[SC]: SecureSession[0x3ffc68b8]: Moving from state 'kEstablishing' --> 'kActive'
I (189030) chip[IN]: CASE Session established to peer: <000000000001B669, 1>
I (189040) chip[IN]: CASE Server enabling CASE session setups
I (189060) chip[EM]: >>> [E:12648r S:25827 M:203024570] (S) Msg RX from 1:000000000001B669 [2742] --- Type 0001:02 (IM:ReadRequest)p[EM]: >>> [E:12647r S:0 M:195917869 (Ack:125362575)] (U) Msg RX from 0:B8E029A9EC613E6E [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)827 M:163819976 (Ack:203024570)] (S) Msg TX to 1:000000000001B669 [2742] --- Type 0001:05 (IM:ReportData)nding msg 163819976 on secure session with LSID: 25827
I (189130) chip[EM]: >>> [E:12648r S:25827 M:203024571 (Ack:163819976)] (S) Msg RX from 1:000000000001B669 [2742] --- Type 0000:10 (SecureChannel:StandaloneAck)
```

```
I (310340) chip[EM]: >>> [E:15053r S:0 M:16731296] (U) Msg RX from 0:E50BD09D33F6F6C2 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1) CASE Server received Sigma1 message . Starting handshake. EC 0x3ffc7d20
I (310360) chip[EM]: <<< [E:15053r S:0 M:125362585 (Ack:16731296)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck) 125362585 to IP address 'UDP:[FD61:3599:E3A1::1C2%st1]:36688'
I (310380) chip[SC]: Received Sigma1 msg
I (310390) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0x0000001234567890
I (310820) chip[EM]: <<< [E:15053r S:0 M:125362586 (Ack:16731296)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:31 (SecureChannel:CASE_Sigma2)sg 125362586 to IP address 'UDP:[FD61:3599:E3A1::1C2%st1]:36688'
I (310840) chip[SC]: Sent Sigma2 msg
I (310870) chip[EM]: >>> [E:15053r S:0 M:16731297 (Ack:125362586)] (U) Msg RX from 0:E50BD09D33F6F6C2 [0000] --- Type 0000:32 (SecureChannel:CASE_Sigma3):0 M:125362587 (Ack:16731297)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck) 125362587 to IP address 'UDP:[FD61:3599:E3A1::1C2%st1]:36688'
I (310900) chip[SC]: Received Sigma3 msg
E (310910) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time11910) chip[DL]: Long dispatch time: 996 ms, for event type 2
[... cut ...]
[2024-02-29-SU-2_8-chip-tool.log](https://github.com/espressif/esp-matter/files/14450773/2024-02-29-SU-2_8-chip-tool.log)
[2024-02-29-TC-SU-2_8-prov1.txt](https://github.com/espressif/esp-matter/files/14450776/2024-02-29-TC-SU-2_8-prov1.txt)
[2024-02-29-TC-SU-2_8-prov2.txt](https://github.com/espressif/esp-matter/files/14450777/2024-02-29-TC-SU-2_8-prov2.txt)

Tools are still running (DUT Timestamp is at 8339650 = 2 hours past), but no OTA Request to Provider 2 up to now. What would be the timeout, after which DUT tries to request the second provider? Or is the test missing something?

PhLuReh, 2024-03-01:
Thank you @shubhamdp for this quick-fix. After 10 minutes, the second provider was queried started downloading the firmware file. So, if there is a requirement from the TH / CSA about the query timeout, it should be set appropriate by defautl.

Is there a way to activate this change from user-code? Or will there fix in the esp-matter-core?

shubhamdp, 2024-03-12:
@PhLuReh I'll see how this can be configured from the application without touching the SDK.

shubhamdp, 2024-04-25:
@PhLuReh Please take a look at this https://github.com/espressif/esp-matter/commit/eea4ed7554c74ceabeee7001367563db787f25f2. I've added an API to configure the timeouts.

closed 2024-04-25 by dhrishi