espressif/esp-matter #816: WDT for Matter node

PhLuReh, 2024-01-30:
**Describe the question/query that you have**
While testing my production ready device, I came to the point, when the CPU simply got stuck. I hoped for a WDT, but it did not fire. Now I found, that the WDT is not enabled in the Matter node - or did I miss the missing piece? How should we enable WDT for Matter node especially in v1.1?

I tried to figure out, how I could access the `TaskHandle_t`. But didn't find the point to add the `esp_task_wdt_reset()`.

Could you please give advise to enable WDT for Matter node?

**Additional context**
...

wqx6, 2024-02-06:
We do not disable WDT by default in ESP Matter examples. Could you check the memuconfig option `ESP_TASK_WDT_EN` in your project?

PhLuReh, 2024-02-06:
Thank you for pointing this out, but I already did enable this menuconfig option (which was not - don't know why). But is the Matter node also using WDT? So can you point out, where the task handle `esp_task_wdt_add()` is called?

shubhamdp, 2024-02-09:
@PhLuReh please take a look at TWDT documentation here: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/wdts.html#task-watchdog-timer-twdt.

Matter task do not explicitly call [`esp_task_wdt_add()`](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/wdts.html#_CPPv416esp_task_wdt_add12TaskHandle_t), but [`esp_task_wdt_init`](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/wdts.html#_CPPv417esp_task_wdt_initPK21esp_task_wdt_config_t) does that for idle task on each CPU. And if the other tasks run prolonged then idle task won't be able to call [`esp_task_wdt_reset()`](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/wdts.html#_CPPv418esp_task_wdt_resetv). And you should get a task watchdog timeout in that case.

dhrishi, 2024-02-27:
@PhLuReh Please close if your questions are answered

PhLuReh, 2024-02-27:
More or less. This may be a solution, but in my opinion, there is a difference between:

1. checking if idle task has time to run (i.e. if Matter task is waiting for some network packets)
2. checking if Matter task is stuck and missing some operations (i.e. sending network packets), but still waiting for some network packets