espressif/esp-matter #817: Regarding analysing wifi packets using wireshark

johnvigg, 2024-01-30:
**Describe the question/query that you have**
Would like to know how to view the Wi-Fi packets using Wireshark and parse it into matter message format.

**Additional context**
I have a setup where a rpi4 acts a chip controller and a couple of ESP32C6 as the matter end device which is running the lighting example. I have a rpi4 with monitor mode running Wireshark and I am able to see the packets but it does not recognize them as matter format and I am not able to understand it. Especially since I am quite new to this. I can see the guideline of the message format in the matter core specification but am not able to compare it. I have attached just a screenshot of Wireshark just for understanding what type of messages I receive but I can send more detailed information if necessary.

Would also like to ask about the encryption of the data? Regarding the encryption on the router end I can disable the password on my router, and I think that will remove the encryption but as far as I understand the matter data itself is also encrypted. Any advice on how I can decrypt would be helpful.

[image]

sayondeep, 2024-01-31:
@johnvigg You can refer [here](https://github.com/project-chip/matter-dissector). On how to configure wireshark for matter protocol. 
I was able to follow the steps and capture the packets while commissioning using CHIP tool.
[image]

jonsmirl, 2024-02-01:
It is far, far easier to print debug logs from both sides of the connection.  Almost everything you will care about is in the debug logs, if some obscure bit is missing, just add an occasional printf to the source code. All of the debug stuff prints before it gets encrypted.

johnvigg, 2024-02-01:
@sayondeep , I tried using the matter-dissector before but I faced some issues. Could you tell me what OS you are currently using and maybe the wireshark version? 

I tried it first on the Kali Linux running on the pi but that failed to run. I tried it again on an ubuntu 22.04 machine and followed all the steps the best I could. I am still new to this but the `sudo apt-get build-dep wireshark` command gave an output saying unable to find a source package for wireshark. other than that, all other commands seem to work fine.

Also at what point do I know if the plugin was installed properly? I have a pcapng file with some matter packets and when I open it in wireshark, it does not recognize as matter packets. Also wireshark seems to already have matter as one of the enabled protocols, could that somehow affect this?

johnvigg, 2024-02-01:
@jonsmirl , that is true, but I want to analyse the packets and see the message format, not to debug but just understand more about how it works: So for me visualising the packets is the best option.

jonsmirl, 2024-02-01:
You are tilting a windmill. Each Matter node has its own private elliptic curve encryption keys (one per fabric) and those keys are inside ARM trustzone or hardware chips. You can probably decode a pair of test nodes talking, but you are not going to get general access to the Matter network. All of the protocol info with lots of diagrams and pictures is in the freely downloadable specification at csa-iot.org. Much, much easier to just read the spec.

sayondeep, 2024-02-02:
@johnvigg I tested it on Ubuntu with wireshark version 3.6.2.
For the `build-dep` command. Please run `software-properties-gtk` and enable `source code`. Also run `sudo apt-get update`. This should fix it on Ubuntu.

Wireshark have added support for matter, see [here](https://gitlab.com/wireshark/wireshark/-/merge_requests/9865). If you are using latest version it may already have that. But it is different from matter-dissector. On version 3.6.2 of wireshark it was not already present. OR 

>all other commands seem to work fine

if you follwed the other steps as you mentioned you have already installed the plugin.If you can see Matter IM Subscription and Matter Exchange under `Analyze->conversation filter` then it should be matter dissector plugin.

johnvigg, 2024-02-02:
@sayondeep , Thank you for the detailed help.
I tried the commands you mentioned and I think I was able to setup the matter-dissector. I also downgraded it to Wireshark 3.6.2, just to be sure.

just few things to mention, when I go to `Analyze->conversation filter` I see matter IM Subscription and Matter Exchange, which I did not see before, so I think the plugin was installed.

Regarding the EAPOL messages I can only see 2 of the messages, namely 1 out of 4 and 3 out of 4. Is this an issue?
I am asking since it still does not recognize as a matter package. I added the wifi password as you mentioned, then I also removed all passwords from the wifi connection and tried. Still does not recognize. 

Also regarding the encryption key mentioned in the github page of matter- dissector would that work for me? If not is there a way to disable encryption on matter? I am just working in the testing phase, so it should be possible right?

Also just to explain my current setup the rpi running Kali Linux captures the packets and then I copy this file to another system where I installed the matter dissector and here I try to parse. I am doing this since I was not able to set it up on Kali.

sayondeep, 2024-02-02:
@johnvigg yes all 4 eapol messages are necessary if you are capturing in monitor mode. Just to be sure that the plugin is working, you can try capturing in managed mode on the same rpi that is running controller.

johnvigg, 2024-02-04:
@sayondeep , Thank you for the quick reply, I tried with the same device (intel nuc) running the controller and the matter-dissector could recognize the different matter packets but the matter data is still encrypted because I have not modified the CHIP_CONFIG_SECURITY_TEST_MODE. I found it in the ESP files but was not able to find it on the controller / chip-tool, could not find it when I searched it online as well. Do you by chance know where to find it?

Regarding the rpi in monitor mode I am still not able to read the packets as matter even though I know the wifi packets are not encrypted. not sure what the difference is...

sayondeep, 2024-02-04:
@johnvigg 
1) For chip-tool you need to change `#define CHIP_CONFIG_SECURITY_TEST_MODE 0`  to `#define CHIP_CONFIG_SECURITY_TEST_MODE 1` in `/path/to/esp-matter/connectedhomeip/connectedhomeip/examples/chip-tool/include/CHIPProjectAppConfig.h` and build the chip-tool again. 
2) For application side Enable `CHIP_CONFIG_SECURITY_TEST_MODE` using `idf.py menuconfig` for the example.
**Note:** In both cases, If you get compilation warning just comment out the line that generates the warning i.e., the line with `#warning` .

sayondeep, 2024-02-04:
@johnvigg regarding capture in monitor mode. If other device is connected to the network can you see other protocols like QUIC,TCP, TLS etc. If you cannot see anything other than 802.11 then wireshark is not able to decrypt the wifi packets.
Can you please try adding your wifi Password like this `<password>:<SSID>` in in Edit->Preferences->Protocols->IEEE 802.11->Decryption keys. Still capture of four-stage WPA/WPA2 handshake is necessary.

johnvigg, 2024-02-04:
@sayondeep , Thank you for all your help. 

I was able to successfully decrypt the matter message after changing the two defines. This was done on the device running the chip-tool controller.
I just want to confirm that this setup provides all the packets that I would have received using the rpi on monitor mode. Since the data sent back from the esp would route back to the controller. Is that correct?

Regarding the rpi in monitor mode I able to see packets from protocols like SSDP, TCP, ARP, IGMPv3. So i think removing the password does remove the encryption.

sayondeep, 2024-02-04:
@johnvigg Yes this setup would contain all packets from controller to end device and from end device to the controller.
Feel free to close the issue if it is resolved.

johnvigg, 2024-02-04:
@sayondeep , Perfect, thank you for your help again.

closed 2024-02-04 by johnvigg