espressif/esp-matter #818: Thermostat Attributes depending on `Heat` and `Cool` features has been removed in release/v1.2 branch (and latest main branch)

hideakitai, 2024-02-02:
**Describe the question/query that you have**
A clear and concise description of what you want to ask.

The following Thermostat attributes, depending on the `Heat` and `Cool` features, have been removed from the `release/v1.1` branch to the `release/v1.2` branch (and still do not exist on the latest main branch). Could you explain why these attributes have been removed? I think these attributes are required for the `Heat` and `Cool` features in Matter v1.2 (not removed or deprecated from spec).

- `Heat`
  - AbsMinHeatSetpointLimit
  - AbsMaxHeatSetpointLimit
  - PIHeatingDemand
  - MinHeatSetpointLimit
  - MaxHeatSetpointLimit
- `Cool`
  - AbsMinCoolSetpointLimit
  - AbsMaxCoolSetpointLimit
  - PICoolingDemand
  - MinCoolSetpointLimit
  - MaxCoolSetpointLimit

**Additional context**
...

commit removed above attributes
https://github.com/espressif/esp-matter/commit/6b60ba75460774910390143ea850b5b7763f777c

release/v1.1
https://github.com/espressif/esp-matter/blob/5abc3095da0da9342dcd1250a01d2c47f840fed0/components/esp_matter/esp_matter_feature.cpp#L654-L670

release/v1.2
https://github.com/espressif/esp-matter/blob/2a5052dc8d7df9b09511d755629f62314ee7658c/components/esp_matter/esp_matter_feature.cpp#L2709-L2720

jadhavrohit924, 2024-02-02:
Hi @hideakitai,
By default esp-matter supports `mandatory` things only, and these attributes are `optional` for `HEAT` and `COOL` thats why these are removed later. If you want these attributes on your device you can add them the same way.

hideakitai, 2024-02-02:
@jadhavrohit924 I got it. Thank you for clarifying.

closed 2024-02-02 by hideakitai