espressif/esp-matter #820: understanding the example esp-now_bridge_light

albrownwood, 2024-02-03:
Hello,
I am trying to experiment with the bridge feature using the espnow_bridge_light example. If I understand this example correctly, after pairing I should see a local light accessory and an empty bridge. The first issue I encounter is that I don't see any local light accessory but I see an empty bridge.

Then I slightly modified the example app_main (see below the add_bridged_thread_entry) to add/remove a switch from the bridge every 2 minutes, the functions are correctly executed as per log but no accessory dynamically appear/disappear in the bridge as I would expect.

Am I missing something? Thanks

```
/*
   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include <esp_err.h>
#include <esp_log.h>
#include <nvs_flash.h>
#include <esp_wifi.h>
#include <esp_matter.h>
#include <esp_matter_console.h>
#include <esp_matter_ota.h>
#include <app_bridged_device.h>
#include <app_priv.h>
#include <app_espnow.h>
#include <app_reset.h>
#include <espnow_bridge.h>

static const char *TAG = "app_main";
uint16_t light_endpoint_id = 0;
uint16_t aggregator_endpoint_id = 0;

using namespace esp_matter;
using namespace esp_matter::attribute;
using namespace esp_matter::endpoint;
using namespace chip::app::Clusters;

static void add_bridged_thread_entry(void* arg)
{
    uint8_t espnow_macaddr[6];
    memset(espnow_macaddr, 1, sizeof(espnow_macaddr));

    while(1){

        ESP_LOGI(TAG, "wait 2 min");
        
        vTaskDelay(120000/portTICK_PERIOD_MS);
        ESP_LOGI(TAG, "Add accesory");

        if(espnow_bridge_match_bridged_switch(espnow_macaddr, 0, ESP_MATTER_ON_OFF_SWITCH_DEVICE_TYPE_ID) != ESP_OK){
            ESP_LOGE(TAG, "error match bridge");
        } else{
            ESP_LOGW(TAG, "success match bridge");
        }

        ESP_LOGI(TAG, "wait 2 min");
        
        vTaskDelay(120000/portTICK_PERIOD_MS);
        
        ESP_LOGI(TAG, "Remove accesory");
        if(espnow_bridge_remove_bridged_switch(espnow_macaddr) != ESP_OK){
            ESP_LOGE(TAG, "error remove bridge");
        } else{
            ESP_LOGW(TAG, "success remove bridge");
        }

    }

}

static void app_event_cb(const ChipDeviceEvent *event, intptr_t arg)
{
    switch (event->Type) {
    case chip::DeviceLayer::DeviceEventType::kServerReady:
    {
        ESP_LOGI(TAG, "Matter node initialized and ready");
        xTaskCreate(add_bridged_thread_entry, "temp_sensor", 4096, NULL, 2, NULL);
        break;
    }
    case chip::DeviceLayer::DeviceEventType::kInterfaceIpAddressChanged:
        ESP_LOGI(TAG, "Interface IP Address changed");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningComplete:
        ESP_LOGI(TAG, "Commissioning complete");
        break;

    case chip::DeviceLayer::DeviceEventType::kFailSafeTimerExpired:
        ESP_LOGI(TAG, "Commissioning failed, fail safe timer expired");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningSessionStarted:
        ESP_LOGI(TAG, "Commissioning session started");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningSessionStopped:
        ESP_LOGI(TAG, "Commissioning session stopped");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningWindowOpened:
        ESP_LOGI(TAG, "Commissioning window opened");
        break;

    case chip::DeviceLayer::DeviceEventType::kCommissioningWindowClosed:
        ESP_LOGI(TAG, "Commissioning window closed");
        break;

    default:
        break;
    }
}

// This callback is invoked when clients interact with the Identify Cluster.
// In the callback implementation, an endpoint can identify itself. (e.g., by flashing an LED or light).
static esp_err_t app_identification_cb(identification::callback_type_t type, uint16_t endpoint_id, uint8_t effect_id,
                                       uint8_t effect_variant, void *priv_data)
{
    ESP_LOGI(TAG, "Identification callback: type: %d, effect: %d", type, effect_id);
    return ESP_OK;
}

// This callback is called for every attribute update. The callback implementation shall
// handle the desired attributes and return an appropriate error code. If the attribute
// is not of your interest, please do not return an error code and strictly return ESP_OK.
static esp_err_t app_attribute_update_cb(attribute::callback_type_t type, uint16_t endpoint_id, uint32_t cluster_id,
                                         uint32_t attribute_id, esp_matter_attr_val_t *val, void *priv_data)
{
    esp_err_t err = ESP_OK;

    if (type == PRE_UPDATE) {
        /* Driver update */
        app_driver_handle_t driver_handle = (app_driver_handle_t)priv_data;
        err = app_driver_attribute_update(driver_handle, endpoint_id, cluster_id, attribute_id, val);
    }

    return err;
}

extern "C" void app_main()
{
    esp_err_t err = ESP_OK;

    /* Initialize the ESP NVS layer */
    nvs_flash_init();

    /* Initialize driver */
    app_driver_handle_t light_handle = app_driver_light_init();
    app_driver_handle_t button_handle = app_driver_button_init();
    app_reset_button_register(button_handle);

    /* Create a Matter node and add the mandatory Root Node device type on endpoint 0 */
    node::config_t node_config;
    node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);

    /* This node handle can be used to create/add other endpoints and clusters. */
    if (!node) {
        ESP_LOGE(TAG, "Matter node creation failed");
    }

    color_temperature_light::config_t light_config;
    light_config.on_off.on_off = DEFAULT_POWER;
    light_config.on_off.lighting.start_up_on_off = nullptr;
    light_config.level_control.current_level = DEFAULT_BRIGHTNESS;
    light_config.level_control.lighting.start_up_current_level = DEFAULT_BRIGHTNESS;
    light_config.color_control.color_mode = (uint8_t)ColorControl::ColorMode::kColorTemperature;
    light_config.color_control.enhanced_color_mode = (uint8_t)ColorControl::ColorMode::kColorTemperature;
    light_config.color_control.color_temperature.startup_color_temperature_mireds = nullptr;
    endpoint_t *endpoint = color_temperature_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
    if (!endpoint) {
        ESP_LOGE(TAG, "Matter color temperature light endpoint creation failed");
    }

    aggregator::config_t aggregator_config;
    endpoint_t *aggregator = endpoint::aggregator::create(node, &aggregator_config, ENDPOINT_FLAG_NONE, NULL);
    if (!aggregator) {
        ESP_LOGE(TAG, "Matter aggregator endpoint creation failed");
    }

    light_endpoint_id = endpoint::get_id(endpoint);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id);

    aggregator_endpoint_id = endpoint::get_id(aggregator);
    ESP_LOGI(TAG, "Switch created with endpoint id %d", aggregator_endpoint_id);

    /* Add additional features to the node */
    cluster_t *cluster = cluster::get(endpoint, ColorControl::Id);
    cluster::color_control::feature::hue_saturation::config_t hue_saturation_config;
    hue_saturation_config.current_hue = DEFAULT_HUE;
    hue_saturation_config.current_saturation = DEFAULT_SATURATION;
    cluster::color_control::feature::hue_saturation::add(cluster, &hue_saturation_config);

    /* Matter start */
    err = esp_matter::start(app_event_cb);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Matter start failed: %d", err);
    }

    err = app_bridge_initialize(node);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Failed to resume the bridged endpoints: %d", err);
    }

    app_espnow_init();

    /* Starting driver with default values */
    app_driver_light_set_defaults(light_endpoint_id);

#if CONFIG_ENABLE_CHIP_SHELL
    esp_matter::console::diagnostics_register_commands();
    esp_matter::console::init();
#endif
}
```

shubhamdp, 2024-02-09:
@albrownwood I hope you followed [additional environment setup](https://github.com/espressif/esp-matter/tree/main/examples/esp-now_bridge_light#1-additional-environment-setup) steps mentioned in the readme.

You probably need three devkits if you want to try:
1. ESP-Now-Switch, It should appear in the the home app after bridging
2. Matter Light, its the standard examples/light, so I do not see any problem this not showing up in the home app.
3. ESP-Now Matter Bridge.
(You can skip the matter light though, if you don't want to test the esp-now-switch and light binding).

IIRC, I tried with chip-tool, followed all the steps and was able to control the light from the esp-now-switch.

bubblesnake, 2024-02-13:
@albrownwood , light endpoint is created on startup even without being commissioned:
`    endpoint_t *endpoint = color_temperature_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
`
You should be able to verify it from the monitor log, or using chip-tool as described in [README:Discovering bridge endpoint](https://github.com/espressif/esp-matter/tree/main/examples/esp-now_bridge_light#21-discovering-bridge-endpoint). Using Google Home I can also see 2 endpoints. However somehow on Apple Home it does not show any.

Your code is also working fine. I can see the ONOFF switch endpoint being created/removed dynamically from log and chip-tool.

albrownwood, 2024-02-13:
@bubblesnake thanks for checking on your side, it looks like I observe the same behavior as you on iOS.
@shubhamdp you mentioned using CHIP tool and the example worked, does the example work on iOS as well on your side?

shubhamdp, 2024-02-14:
@albrownwood I have only tested with chip-tool.

sayondeep, 2024-02-23:
@albrownwood 
I added the following piece of code to test the behaviour of bridge on Apple Home.
```
    err = app_bridge_initialize(node,create_bridge_devices);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Failed to resume the bridged endpoints: %d", err);
    }

    uint8_t espnow_macaddr[6];
    memset(espnow_macaddr, 1, sizeof(espnow_macaddr));
    app_bridged_device_t *bridged_device = app_bridge_create_bridged_device(node, aggregator_endpoint_id, ESP_MATTER_ON_OFF_LIGHT_DEVICE_TYPE_ID,
                                           ESP_MATTER_BRIDGED_DEVICE_TYPE_ESPNOW,
                                           app_bridge_espnow_address(espnow_macaddr,0), NULL);
    if (bridged_device == NULL) {
        ESP_LOGE(TAG, "Failed to create bridged device (on_off light)");
    }

        ESP_LOGI(TAG, "Create/Update bridged node for " MACSTR " bridged device on endpoint %d", MAC2STR(espnow_macaddr),
                app_bridge_get_matter_endpointid_by_espnow_macaddr(espnow_macaddr));
```
I was able to see the local light and a bridged light as expected. But,
If I replace the ESP_MATTER_ON_OFF_LIGHT_DEVICE_TYPE_ID with ESP_MATTER_ON_OFF_SWITCH_DEVICE_TYPE_ID , I was not able to see any device after successful commissioning. 
ON_OFF_SWITCH_DEVICE_TYPE being not supported by Apple Home is a possible reason for this behaviour.

albrownwood, 2024-02-25:
@sayondeep thanks for the comment, I think you nailed it, I also see the accessory appearing/disappearing as expected on iOS Home when I replace the accessory type from ESP_MATTER_ON_OFF_SWITCH_DEVICE_TYPE_ID to ESP_MATTER_ON_OFF_LIGHT_DEVICE_TYPE_ID. So yes it looks like the on_off switch does not appear on iOS if part of a bridge.
@shubhamdp is there anything that can be done on esp-matter about that or the fix is all on iOS side?

dhrishi, 2024-02-27:
@albrownwood I don't think there's anything that can be done on the device firmware side.

albrownwood, 2024-02-27:
ok I am closing this, thanks.

closed 2024-02-27 by albrownwood