espressif/esp-matter #823: How to specify number of LEDs for ws8212 device

lmatter, 2024-02-04:
I'm trying to build the light example for an ESP32C3 devkit M with similar functionality to the zerocode led_strip configuration.  I see that I can change the GPIO pin in the led_driver_config_t  in device.c, but I'm not sure how to specify how many LEDs should be lit.  Is there any other documentation for this other than the dev guide?    

**Additional context**
Obviously I'm new to this.  I was able to get the sample working on my device and added to Home Assistant.   As discussed in https://github.com/espressif/esp-matter/issues/312 I am unable to change the color of the LED, but brightness works fine.  Any other helpful tips or suggestions for where to learn more are greatly appreciated.

Btw, is the Zerocode ezc code open source?  I can't find it.

shubhamdp, 2024-02-09:
@lmatter In `examples/generic_switch` we have some example demonstration for to showcase [multiple buttons](https://github.com/espressif/esp-matter/blob/main/examples/generic_switch/main/app_main.cpp#L121). Please take a look at that.

Also, there is [led_indicator](https://github.com/espressif/esp-iot-solution/tree/master/components/led/led_indicator) component which supports led_strip, high level driver code for your application.

At this moment, zerocode is closed source. Let me know if you need more help.

dhrishi, 2024-03-02:
Closing as the questions are answered

closed 2024-03-02 by dhrishi