espressif/esp-matter #824: ESP32 BLE Mesh Bridge example crashing

tomlinyama, 2024-02-05:
The BLE Mesh Bridge example crashes upon startup and then restarts.
[image]
**Environment**
- ESP-Matter Commit Id: 52a95ea
- ESP-IDF Commit Id: 6b1f40b9bf
- SoC (eg: ESP32 or ESP32-C3): ESP32-C3
- Device Logs (Please attach the log file):
- Host Machine OS: Ubuntu 22.04

wqx6, 2024-02-06:
Looks like you didn't set the DeviceInstanceInfoProvider for in your code. Are you using custom device instance info provider in your example? if yes, please call `esp_matter::set_custom_device_instance_info_provider()` before your call `esp_matter::start()`.

tomlinyama, 2024-02-06:
The example hasn't been modified in any way. If the external platform is removed, it won't restart.

chshu, 2024-02-07:
@tomlinyama The issue is fixed by this commit: https://github.com/espressif/esp-matter/commit/668e22da6fca1092e43fbf8fe615db71b9b78e2d, please try again.