espressif/esp-matter #825: All device types app not building properly

FloBer431, 2024-02-05:
No matter how often I try to build the All_Device_types_app example, it keeps on failing the build.
I'm building in VSCode under Windows 11, connected to WSL2.
The whole building process starts properly, but shortly before the end, an error occures.

From here the FAILED exceptions starts and stopping the building process.

` [1264/1278] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/esp_matter_console_helpers.cpp.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/esp_matter_console_helpers.cpp.obj 
/home/ber/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM -DIDF_VER=\"v5.3-dev-1353-gb3f7e2c8a4\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -Iconfig -I../main -I/home/ber/esp/esp-idf/components/newlib/platform_include -I/home/ber/esp/esp-idf/components/freertos/config/include -I/home/ber/esp/esp-idf/components/freertos/config/include/freertos -I/home/ber/esp/esp-idf/components/freertos/config/xtensa/include -I/home/ber/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/ber/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/ber/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -I/home/ber/esp/esp-idf/components/freertos/esp_additions/include -I/home/ber/esp/esp-idf/components/esp_hw_support/include -I/home/ber/esp/esp-idf/components/esp_hw_support/include/soc -I/home/ber/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/ber/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/home/ber/esp/esp-idf/components/heap/include -I/home/ber/esp/esp-idf/components/log/include -I/home/ber/esp/esp-idf/components/soc/include -I/home/ber/esp/esp-idf/components/soc/esp32 -I/home/ber/esp/esp-idf/components/soc/esp32/include -I/home/ber/esp/esp-idf/components/hal/platform_port/include -I/home/ber/esp/esp-idf/components/hal/esp32/include -I/home/ber/esp/esp-idf/components/hal/include -I/home/ber/esp/esp-idf/components/esp_rom/include -I/home/ber/esp/esp-idf/components/esp_rom/include/esp32 -I/home/ber/esp/esp-idf/components/esp_rom/esp32 -I/home/ber/esp/esp-idf/components/esp_common/include -I/home/ber/esp/esp-idf/components/esp_system/include -I/home/ber/esp/esp-idf/components/esp_system/port/soc -I/home/ber/esp/esp-idf/components/esp_system/port/include/private -I/home/ber/esp/esp-idf/components/xtensa/esp32/include -I/home/ber/esp/esp-idf/components/xtensa/include -I/home/ber/esp/esp-idf/components/xtensa/deprecated_include -I/home/ber/esp/esp-idf/components/lwip/include -I/home/ber/esp/esp-idf/components/lwip/include/apps -I/home/ber/esp/esp-idf/components/lwip/include/apps/sntp -I/home/ber/esp/esp-idf/components/lwip/lwip/src/include -I/home/ber/esp/esp-idf/components/lwip/port/include -I/home/ber/esp/esp-idf/components/lwip/port/freertos/include -I/home/ber/esp/esp-idf/components/lwip/port/esp32xx/include -I/home/ber/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/ber/esp/esp-idf/components/lwip/port/esp32xx/include/sys -I/home/ber/esp-matter/device_hal/device/include -I/home/ber/esp-matter/device_hal/led_driver/include -I/home/ber/esp/esp-idf/components/driver/deprecated -I/home/ber/esp/esp-idf/components/driver/i2c/include -I/home/ber/esp/esp-idf/components/driver/touch_sensor/include -I/home/ber/esp/esp-idf/components/driver/twai/include -I/home/ber/esp/esp-idf/components/driver/touch_sensor/esp32/include -I/home/ber/esp/esp-idf/components/esp_pm/include -I/home/ber/esp/esp-idf/components/esp_ringbuf/include -I/home/ber/esp/esp-idf/components/esp_driver_gpio/include -I/home/ber/esp/esp-idf/components/esp_driver_pcnt/include -I/home/ber/esp/esp-idf/components/esp_driver_gptimer/include -I/home/ber/esp/esp-idf/components/esp_driver_spi/include -I/home/ber/esp/esp-idf/components/esp_driver_mcpwm/include -I/home/ber/esp/esp-idf/components/esp_driver_ana_cmpr/include -I/home/ber/esp/esp-idf/components/esp_driver_i2s/include -I/home/ber/esp/esp-idf/components/esp_driver_sdmmc/include -I/home/ber/esp/esp-idf/components/sdmmc/include -I/home/ber/esp/esp-idf/components/esp_driver_sdspi/include -I/home/ber/esp/esp-idf/components/esp_driver_sdio/include -I/home/ber/esp/esp-idf/components/esp_driver_dac/include -I/home/ber/esp/esp-idf/components/esp_driver_rmt/include -I/home/ber/esp/esp-idf/components/esp_driver_tsens/include -I/home/ber/esp/esp-idf/components/esp_driver_sdm/include -I/home/ber/esp/esp-idf/components/esp_driver_i2c/include -I/home/ber/esp/esp-idf/components/esp_driver_uart/include -I/home/ber/esp/esp-idf/components/vfs/include -I/home/ber/esp/esp-idf/components/esp_driver_ledc/include -I/home/ber/esp/esp-idf/components/esp_driver_parlio/include -I/home/ber/esp/esp-idf/components/esp_driver_usb_serial_jtag/include -I../managed_components/espressif__button/include -I/home/ber/esp/esp-idf/components/esp_adc/include -I/home/ber/esp/esp-idf/components/esp_adc/interface -I/home/ber/esp/esp-idf/components/esp_adc/esp32/include -I/home/ber/esp/esp-idf/components/esp_adc/deprecated/include -I/home/ber/esp-matter/components/esp_matter -I/home/ber/esp-matter/components/esp_matter/utils -I/home/ber/esp-matter/connectedhomeip/connectedhomeip/zzz_generated/app-common -I/home/ber/esp-matter/connectedhomeip/connectedhomeip/third_party/nlfaultinjection/include -I/home/ber/esp-matter/connectedhomeip/connectedhomeip/src -I/home/ber/esp-matter/components/esp_matter/zap_common -I/home/ber/esp-matter/connectedhomeip/connectedhomeip/src/platform/ESP32 -I/home/ber/esp-matter/connectedhomeip/connectedhomeip/src/platform/OpenThread -I/home/ber/esp-matter/connectedhomeip/connectedhomeip/src/include -I/home/ber/esp-matter/connectedhomeip/connectedhomeip/src/lib -I/home/ber/esp-matter/connectedhomeip/connectedhomeip/examples/platform/esp32 -I/home/ber/esp-matter/connectedhomeip/connectedhomeip/third_party/nlassert/repo/include -I/home/ber/esp-matter/connectedhomeip/connectedhomeip/third_party/nlio/repo/include -Iesp-idf/chip/src/include -Iesp-idf/chip/include -Iesp-idf/chip/gen/include -I/home/ber/esp-matter/connectedhomeip/connectedhomeip/config/esp32 -I/home/ber/esp-matter/connectedhomeip/connectedhomeip/config/esp32/../.. -I../managed_components/espressif__esp_encrypted_img/include -I../managed_components/espressif__esp_insights/include -I../managed_components/espressif__esp_diagnostics/include -I/home/ber/esp/esp-idf/components/bt/include/esp32/include -I/home/ber/esp/esp-idf/components/bt/common/osi/include -I/home/ber/esp/esp-idf/components/bt/common/api/include/api -I/home/ber/esp/esp-idf/components/bt/common/btc/profile/esp/blufi/include -I/home/ber/esp/esp-idf/components/bt/common/btc/profile/esp/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ans/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/bas/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/dis/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gap/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gatt/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/hr/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/htp/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ias/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ipss/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/lls/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/prox/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/cts/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/tps/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/hid/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/sps/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/util/include -I/home/ber/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/include -I/home/ber/esp/esp-idf/components/bt/hos [... cut ...]
   20 | void esp_vfs_dev_uart_use_driver(int uart_num) __attribute__((deprecated("Please use uart_vfs_dev_use_driver() instead")));
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
[1268/1278] Building CXX object esp-idf/esp_matter/CMakeFiles/__idf_esp_matter.dir/home/ber/esp-matter...omeip/connectedhomeip/src/app/clusters/time-synchronization-server/time-synchronization-server.cpp.obj
ninja: build stopped: subcommand failed.

 *  The terminal process "/bin/bash '-c', 'ninja '" terminated with exit code: 1. `
Can anybody help me?
Also the create --device_type command does not work.

jadhavrohit924, 2024-02-05:
@FloBer431 Which esp-idf version you are using?

FloBer431, 2024-02-05:
> @FloBer431 Which esp-idf version you are using?

I think it's V5.1.2. where can I check it?

jadhavrohit924, 2024-02-05:
Esp-matter is compatible with v5.1.2. Can you please share me the esp-idf commit, esp-matter commit and the steps that you have followed, patch of your local changes, if any?

FloBer431, 2024-02-05:
> Esp-matter is compatible with v5.1.2. Can you please share me the esp-idf commit, esp-matter commit and the steps that you have followed, patch of your local changes, if any?

Sorry I'm pretty new working with these system, where do I find the commit?
So step by step:
- Starting VSCode and connecting to WSL
- opening the example project
- connecting com-port with wsl over powershell
- setting board to esp32
- starting to build 
- getting the error i posted before

I didn't make any changes.

jadhavrohit924, 2024-02-06:
@FloBer431 Use the command `git log` and check the commit id of `HEAD`

FloBer431, 2024-02-06:
> @FloBer431 Use the command `git log` and check the commit id of `HEAD`

That doesn't work for me, as I have no git repositorys `fatal: not a git repository (or any of the parent directories): .git`

FloBer431, 2024-02-06:
I don't really know why, but after some restart of VSCode, the WSL connection and various tries, it worked ...

Thanks

closed 2024-02-06 by FloBer431