espressif/esp-matter #827: Errors building my first own ESP-Matter project

FloBer431, 2024-02-06:
Hello there,

I'm currently building my first own project with ESP-Matter.
The device is going to be an contact sensor device.
I reffered to a former issue: https://github.com/espressif/esp-matter/issues/649 where some code was provided for building a contact sensor device.
After some struggles I thought i managed to make it work, but I'm facing following problem:

`-- Configuring done
CMake Error:
  Error evaluating generator expression:

    $<TARGET_FILE:__idf_main>

  Target "__idf_main" is not an executable or library.

CMake Error:
  Error evaluating generator expression:

    $<TARGET_FILE:__idf_main>

  Target "__idf_main" is not an executable or library.

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.`

I think I made an error somewhere in my app_driver.ccp, app_main.cpp or app_priv.h, maybe somebody can help me.

I'm working on a Windows 11 system, in VSCode with WSL connection.

sayondeep, 2024-02-07:
@FloBer431 the error could be in the projects main component CMakeLists.txt . Please refer [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html) for how ESP-IDF build system works.

dhrishi, 2024-02-16:
@FloBer431 Please close this if the issue is resolved

dhrishi, 2024-02-27:
Closing this. Please re-open in case of any problems

closed 2024-02-27 by dhrishi