espressif/esp-matter #828: Matter over Ethernet on ESP Thread BR

supersunil, 2024-02-06:
Hi,
I am trying to implement Matter over Ethernet on [ESP Thread BR](https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html), to do so, I started with changing `sdkconfig`

```
CONFIG_ENABLE_WIFI_TELEMETRY=n
CONFIG_ENABLE_ETHERNET_TELEMETRY=y
```

When I tried building the same, got hit with,

```
../../../../../../esp-matter-v1.2/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/NetworkCommissioningDriver_Ethernet.cpp: In member function 'virtual CHIP_ERROR chip::DeviceLayer::NetworkCommissioning::ESPEthernetDriver::Init(chip::DeviceLayer::NetworkCommissioning::Internal::BaseDriver::NetworkStatusChangeCallback*)':
../../../../../../esp-matter-v1.2/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/NetworkCommissioningDriver_Ethernet.cpp:56:5: error: 'eth_esp32_emac_config_t' was not declared in this scope; did you mean 'eth_mac_config_t'?
   56 | 	eth_esp32_emac_config_t esp32_emac_config = ETH_ESP32_EMAC_DEFAULT_CONFIG();
  	| 	^~~~~~~~~~~~~~~~~~~~~~~
  	| 	eth_mac_config_t
../../../../../../esp-matter-v1.2/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/NetworkCommissioningDriver_Ethernet.cpp:57:5: error: 'esp32_emac_config' was not declared in this scope
   57 | 	esp32_emac_config.smi_mdc_gpio_num    	= CONFIG_ETH_MDC_GPIO;
  	| 	^~~~~~~~~~~~~~~~~
../../../../../../esp-matter-v1.2/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/NetworkCommissioningDriver_Ethernet.cpp:59:49: error: 'esp_eth_mac_new_esp32' was not declared in this scope
   59 | 	esp_eth_mac_t * mac                   	= esp_eth_mac_new_esp32(&esp32_emac_config, &mac_config);
  	|                                             	^~~~~~~~~~~~~~~~~~~~~
At global scope:
cc1plus: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier diagnostics
[337/454] c++ obj/third_party/connectedhomeip/src/protocols/user_directed_commissioning/libChipProtocols.UserDirectedCommissioningClient.cpp.oninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
ninja failed with exit code
```
I do understand that the Ethernet chipset which is used on [ESP Thread BR](https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html) is different from [ESP-ETHERNET-KIT](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-ethernet-kit.html), do I need to specify additional configuration to make it compatible? 

Thanks in advance

shubhamdp, 2024-02-08:
Please help with following details:

**Describe the bug**
A clear and concise description of what the bug is.

**Environment**
- ESP-Matter Commit Id:
- ESP-IDF Commit Id:
- SoC (eg: ESP32 or ESP32-C3):
- Example you are building:
- Build Logs (Please attach the log file):
- Host Machine OS:
- 
**Any additional details**
...

supersunil, 2024-02-09:
# Query for Matter over Ethernet Update

ESP Matter - [release/v1.2](https://github.com/espressif/esp-matter/tree/release/v1.2)
ESP IDF - [release/5.1.2](https://github.com/espressif/esp-idf/tree/v5.1.2)
Example - [Matter Light](https://github.com/espressif/esp-matter/tree/release/v1.2/examples/light)
Board - [ESP Thread Border Router/Zigbee Gateway v1.2](https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html#ethernet-based-thread-border-router)
Host - MacOs  v14.3

Updated `sdkconfig.defaults`

```
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y

CONFIG_ENABLE_WIFI_TELEMETRY=n
CONFIG_ENABLE_ETHERNET_TELEMETRY=y

#
# Etherenet
# 
CONFIG_EXAMPLE_CONNECT_WIFI=n
CONFIG_EXAMPLE_CONNECT_ETHERNET=y
CONFIG_EXAMPLE_ETHERNET_EMAC_TASK_STACK_SIZE=2048
CONFIG_EXAMPLE_USE_SPI_ETHERNET=y
CONFIG_EXAMPLE_USE_W5500=y
CONFIG_EXAMPLE_ETH_SPI_HOST=2
CONFIG_EXAMPLE_ETH_SPI_SCLK_GPIO=21
CONFIG_EXAMPLE_ETH_SPI_MOSI_GPIO=45
CONFIG_EXAMPLE_ETH_SPI_MISO_GPIO=38
CONFIG_EXAMPLE_ETH_SPI_CS_GPIO=41
CONFIG_EXAMPLE_ETH_SPI_CLOCK_MHZ=36
CONFIG_EXAMPLE_ETH_SPI_INT_GPIO=39
CONFIG_EXAMPLE_ETH_PHY_RST_GPIO=40
CONFIG_EXAMPLE_ETH_PHY_ADDR=1
# end of Etherenet
```
Here is the [build_error.log](https://github.com/espressif/esp-matter/files/14217779/build_error.log)

After doing some searching, I found [ethernet-config](https://github.com/espressif/connectedhomeip/blob/6347d0839c6cc94e5a20ab80821679bc2e2522d1/src/platform/ESP32/NetworkCommissioningDriver_Ethernet.cpp#L45) states currently supports ethernet IP101. To make it work with SPI ethernet modules, can I override the same?

If so, how can I override `CHIP_ERROR ESPEthernetDriver::Init(NetworkStatusChangeCallback * networkStatusChangeCallback)` in `main.cpp` of the project?

shubhamdp, 2024-02-09:
You have to extend the `ESPEthernetDriver` and implement the `Init()` method from there.
https://github.com/espressif/connectedhomeip/blob/6347d0839c6cc94e5a20ab80821679bc2e2522d1/src/platform/ESP32/NetworkCommissioningDriver.h#L178

Then you can initialize the driver in [esp_matter_core.cpp](https://github.com/espressif/esp-matter/blob/release/v1.2/components/esp_matter/esp_matter_core.cpp#L915-L921)

Right now we do not have a provision to configure this. Please change the esp_matter sdk to test what you are doing. We'll make some arrangements to specify the network commissioning driver.

closed 2024-02-14 by supersunil