espressif/esp-matter #829: symbol SEC_CERT_DAC_PROVIDER kconfig error

jonsmirl, 2024-02-06:
Not sure what is going on here...
~~~
warning: default on the choice symbol SEC_CERT_DAC_PROVIDER (defined at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:860, /home/jonsmirl/aosp/esp-matter/components/esp_matter/Kconfig:48) will have no effect, as defaults do not affect choice symbols
warning: the choice symbol SEC_CERT_DAC_PROVIDER (defined at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:860, /home/jonsmirl/aosp/esp-matter/components/esp_matter/Kconfig:48) is defined with a prompt outside the choice
~~~

shripad621git, 2024-02-07:
@jonsmirl , thanks for reporting , we will add the fix soon.

shubhamdp, 2024-02-09:
`SEC_CERT_DAC_PROVIDER` config name is used at two places, ([esp-matter](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/Kconfig#L48-L51), [connectedhomeip](https://github.com/project-chip/connectedhomeip/blob/master/config/esp32/components/chip/Kconfig#L874-L879)) and that why this conflict warning.

shripad621git, 2024-05-15:
@jonsmirl , I revisted the changes for the config options and had the observations mentioned below:
The config option `SEC_CERT_DAC_PROVIDER ` is used in Kconfig of esp-matter repository to help users to select all the `DAC_PROVIDER_OPTIONS` at a go. Renaming the option creates an additional overhead and confusion for users to set two config options for the same functionality.
Since the warning does not affect the overall functionality, hence, for the ease of use , we have kept the name of the config options in esp-matter and connectedhomeip the same.

closed 2024-05-28 by jonsmirl